Marus/cortex-debug #1: How to Support or Donate?

madcowswe, 2018-01-17:
Hi, I saw this project over on [Reddit](https://www.reddit.com/r/embedded/comments/7q1hri/visual_studio_code_extension_for_debugging_arm/). I really like the initiative and I want to support your work. Is there a way to send a monetary contribution?

Marus, 2018-01-17:
Thanks for the interest. At the moment I don’t have anything setup for donations, honestly wasn’t something I had considered until now.

If I do set something in the future I’ll post about it here.

And of course any feedback you might have is also valuable; either issues you encounter, or features you’d like to see.

adammunich, 2019-01-01:
Just commenting to say that what you have so far is great and you should keep up the good work. The big companies have really reached a point of no further progress on this subject.

julian-poidevin, 2019-10-15:
@Marus Is there any donation system now, I'd really like to support the extension development !

madcowswe, 2021-02-14:
Hi @Marus let me know if there is any way we can support this work. If you are open to some paid support or sponsoring of features?
Thanks!

diggit, 2021-05-24:
Hi @Marus and @haneefdm,
I'd also like to financially support development of this extension, on regular basis. I believe, there is more of us...
Librepay/Bountysource/Patreon/Github or some other platform could do the job.
What do you think?

madcowswe, 2021-11-30:
Hi,
Thanks again for the great work on this.
Since you asked if there was any features I'd like to see, I do have one now:

I would like to be able to debug with the disassembly. Basically a special tab that shows the disassembly at the currently paused location, even better if we can set breakpoints and step there too.
There is a feature "show disassembly of this function" right now, but that's not quite enough.

Let me know if I can "vote with my wallet" to get this prioritized?

Thanks again for your great work.

haneefdm, 2021-12-01:
Thank you all for the offers. I can't speak for @Marus, but at this point, I cannot accept cash donations because of my other job.

What I would be thankful for though is time. To answer questions new users typically have, documentation (Wiki), FAQs. Especially with stuff you are currently using or have used before. I don't have access to devices, boards, SW environments, so it is challenging. I don't even have the physical space for all that.