Marus/cortex-debug #7: svdFile not loading

lucasholzhaeuer, 2018-01-26:
Hello, really enjoying your plugin. For whatever reason, the SVD file I have is not loading. I have a link in the configuration to the svd file as shown below. The SVD file is for the nrf52832_xxAA board from Nordic Semiconductors. Their SDK with the SVD file can be found at [http://developer.nordicsemi.com/nRF5_SDK/nRF5_SDK_v14.x.x/](url)

`            "svdFile": "C:/projects/svd/nrf52.svd",`

Thoughts?

Marus, 2018-01-27:
I’ll try to have a look in the next few days and see if I can see why it isn’t parsing that one properly. There are some variations in what different companies use in their SVD files, so possible that Nordic is using some feature or combination of features that I don’t parse correctly currently, but haven’t encountered it in the SVD for any of the chips I currently use in my projects (mostly STM32 and some Kinetis parts - all of which do seem to work).

Marus, 2018-01-27:
I just had a quick look visual scan through the SVD file; I see what I expect the issue with parsing it likely is. If it is only the one I’ve spotted by looking through the SVD file then it shouldn’t be too hard to fix. I will try to push an update Sunday night or Monday to hopefully fix your issue. Sorry for the inconvenience.

Marus, 2018-01-29:
Just to keep you updated on this. I've looked into the parsing of the nrf52.svd file there were actually three parts that it couldn't currently parse:

- There was an alternate representation of the bits for a field within a register that I wasn't parsing correctly (why the SVD specification has three different ways to represent the exact same information I don't know). From the examples I can find this issue would effect SVD files from Nordic, Fujitsu, and Spansion.
- There was a slight variation of repeating register definitions I hadn't handled properly. From the examples I can find this issue would effect SVD files from Nordic, Microchip (Atmel), and some of NXP's LPC line (but not all)
- They make use of a "cluster" feature - allowing them to make sub-groups of registers within a peripheral (and optionally repeat those subgroups). From the examples I can find this would effect only Nordic and Microchip (Atmel) parts.

I have fixes prepared for the first two points - I expect I will release these tonight. 

With just the first two fixes I expect you will have the following results with the nrf52.svd:

- The following peripherals will display correctly: AAR, BPROT, CCM, CLOCK, COMP, ECB, EGU0, EGU1, EGU2, EGU3, EGU4, EGU5, FPU, GPIOTE, LPCOMP, NVMC, P0, RADIO, RNG, RTC0, RTC1, RTC2, SWI0, SWI1, SWI2, SWI3, SWI4, SWI5, TEMP, TIMER0, TIMER1, TIMER2, TIMER3, TIMER4, TWI0, TWI1, UART0, UICR, WDT
- The following peripherals will have some registers missing: FICR, I2S, MWU, NFCT, PDM, POWER, PPI, PWM0, PWM1, PWM2, QDEC, SAADC, SPI0, SPI1, SPI2, SPIM0, SPIM1, SPIM2, SPIS0, SPIS1, SPIS2, TWIM0, TWIM1, TWIS0, TWIS1, UARTE0

I will be working on a fix for the remaining issue to get everything complete for the NRF51 and 52 (as well as the Microchip (Atmel) ARM chips), but this might take me a few days as this is a more significant change than the first two.

Marus, 2018-01-30:
I've released v0.1.11 (may take a while before it shows in the marketplace). I've revamped the SVD parsing in this release, and I think it should fix all three issues I had noted above that were not being handled in the nrf52.svd file. Since I do not have an NRF52 board I cannot test to make sure it is interpreting everything 100% correctly; please let me know of any suspicious behaviour and I will attempt to address the issues.

closed 2018-01-30 by Marus

lucasholzhaeuer, 2018-01-30:
It looks good, I briefly loaded it up and they're all showing. I'll let you know if any other problems arise. Thank you for your quick response!

Samonitari, 2018-01-30:
[LPC82x.svd.zip](https://github.com/Marus/cortex-debug/files/1678869/LPC82x.svd.zip)
Hello @Marus ,

Thanks for the plugin! I just found it today on reddit, and it is awesome!
Sorry for jumping in a closed issue, but I have the same problem - my SVD file for LPC82x series uC does not load.
It is also used for rust device crate generation, and it works for that purpose - but the second issue you mentioned is present - some registers are duplicated, for dual purpose.

If it isn't much to ask, could you look at it?  I am using plugin version 0.1.11 from marketplace.
Best regards,
Krisztián.

Marus, 2018-01-31:
Hi @Samonitari,

Thanks. The extension is still new, and has a few rough edges, but I think it has the potential to be great, and hope people enjoy using it.

I just released a V0.1.12 to the marketplace (may take a little while before it shows up for you to install as an update). I think this will fix the issues with the LPC82x.svd file you attached. Please give that a try and let me know if it is working properly.

Also, if you notice any other issues it is best to open a new issue on GitHub, as it can be easy to miss ones added on to other issues (especially one that was marked closed).

Samonitari, 2018-01-31:
Hi @Marus ,

I already tested it - although not thoroughly - and it works! Yay!
I'll report issues separately next time. It was the same wrong behaviour - but I guess for different technical reason - than the one described in the issue, and I saw your recent activity, so I was hoping it will get noticed - and it did!

## Offtopic
I am just trying out Rust for cortex-m (have some C experience there, but I am working as desktop C++ dev).
I am not a big guy, but I'll gladly report anything that hits my basic usage.
Niche toolchains like the one I am using especially needed this plugin, as there is no official IDE out there. VS Code is the closesest to declaring full support for Rust, and I did not want the monstrous Eclipse again, so thanks for your project again. I will mention your work to a few people developing the embedded Rust, maybe they haven't spotted it!

Marus, 2018-01-31:
Hey @Samonitari,

Glad that SVD fix seems to be working. The SVD specification is kind of strange in places, with multiple ways to represent the exact same thing - and different companies tending to use different representations. So it's sometimes easy to miss one particular variation - especially if it is only used by one of the companies that you don't use a part from regularly (I mostly use STM32 parts, and a couple of the Kinetis parts).

Regarding Rust support - I'm going to start another issue for discussions about that. It is something that I want to support with the extension (currently it isn't though), as it is something I to experiment with myself.