Marus/cortex-debug #9: Debug arguments not recognized

suphaWiz, 2018-01-29:
Hi!
Really like your extension so far! However, I am having problems adding my own debug arguments.
Specifically, I am trying to enable monitor mode debugging on the nrf52 platform, and therefore need to set the "mon exec SetMonModeDebug = 1" and "mon exec SetMonModeVTableAddr = 0x23000" arguments. I get some error stating "No such file or directory". Can also mention that these arguments are recognized perfectly fine by the code-debug/Native debug extension.

Marus, 2018-01-30:
I expect you are trying to add these to the debuggerArgs property in the launch.json - which would not be correct, that property is for more command line arguments. 

At the moment there isn’t an option to automatically run commands at launch (you can manually run them in the debugger output view). Wasn’t really an intentional choice to not allow it, just something that isn’t in place (as in most cases the needed commands are already run).

I will add a new property “postLaunchCommands” for gdb commands to be executed after the automatic launch sequence. This will be in the next update (likely within the next few hours; just need to do some final testing).

Marus, 2018-01-30:
I've released a v0.1.11 update (may take a little while before it shows up in the marketplace). This adds the new "postLaunchCommands" property to the launch.json configuration. Let me know if everything works properly when your two commands are added there.

suphaWiz, 2018-01-30:
I have tested the new version and I can confirm that the lunch commands are processed properly. I'm closing the issue. Thanks a lot!

closed 2018-01-30 by suphaWiz