Marus/cortex-debug #12: Rust Support

Marus, 2018-01-31:
This is an issue to discuss things related to support for Rust.

Samonitari, 2018-01-31:
Hi @Marus!

With [**Rust**](https://marketplace.visualstudio.com/items?itemName=kalitaalexey.vscode-rust) extension, and custom build task using [**xargo**](https://github.com/japaric/xargo), basic debugging is working. I mean, like, source line is displayed correctly on debug hitting breakpoints, step-in/out/over is okay - guess that is language independent mostly. Register view of course works, as it uses the separately supplied SVD.
Variable view is definitly lacking thou. I'll experiment tomorrow more!

Marus, 2018-01-31:
Well - in the current released version of my extension I don't have rust breakpoints enabled - that part is an easy fix. There is also a few issues with my code that would initialize the ITM, DWT, and TPIU registers if you are using SWO decoding in the IDE (had to explicitly set c syntax for expressions in GDB while I was setting up those debug devices, as it would auto to rust - and rust has a different operator for bitwise inversion than c).

I'll push an update later today with those fixes; and then start looking into other improvements (for example I expect the disassembly view likely doesn't work well at all with rust; and as you mentioned the variables view). Hopefully over the next while we can make it into a good environment for debugging rust.

Marus, 2018-01-31:
I've just released v0.1.13 to the marketplace - this has a few changes all related to better supporting rust within the extension:

* Allow setting breakpoints in rust code
* Fixes for the ITM decoding support; the GDB configuration scripts will work properly, and the console decoder will work with the format that iprintln! sends things (up to 4 characters at a time; most c implementations I've seen only do 1 at a time) - this wouldn't help you with the LPC82x though as the Cortex-m0 doesn't support SWO/ITM output

I'm going to keep this issue open as a place to discuss further improvements to support the rust ecosystem within the Cortex-Debug extension.

Samonitari, 2018-02-07:
I did some checking the last days in my free time - but as you wrote ITM is out of my reach, because I don't own currently any non-m0[+] chip.

Breakpoints are working, but needs a bit more testing: there was some funny behaviour I observed, not sure if it is related to the extension, or the timing of my interrupts.
Feedback coming this weekend!

perlindgren, 2018-02-11:
Great work!
I'm teaching a course in embedded systems design (at Luleå University of Technology Sweden), where the student are for the first time using our RTFM framework in Rust for embedded programming. (using the ST-Nucleo64 as well as their own pcbs which they design during the course.) 

We use vscode/rls as the suggested programming environment, with the "Native Debug" plugin. I have been playing around with your plugin, as a potential replacement. Looks promising so far, however I did not get swo/itm tracing working. (Not sure how exactly the launch config should look.) Moreover, tracing is done by compacting up to 4 chars in a single package (reducing OH in comparison to single char approach taken by e.g., CMSIS). To that end we use `itmdump` running in a console. (That I got working together with your plugin, but not the internal decoder running in vscode.)

A few example launch configs could be a great help, as well as some pointer as where to implement a decoder extension for the itm packed format (if necessary). 

Best regards,
  Per Lindgren

perlindgren, 2018-02-12:
Thanks Marcus, itm trace works like a charm (at least for the small test I did).
I used 1 mbit on openocd and nucleo 64 (stlinkv2-1). Target running at 16 mhz.
Also i tried semihosting, it works as well, you get the output in "adapter output". However i had to manually enable semihosting in the gdb console (else the "semihosting" breakpoint is not caught correctly),

A few questions/suggestions. 

0 - an option for semihosting (if there is not one already)

0.5  - perhaps an after launch script is even better where you just give the "monitor arm semihosting enable"

1 - openocd has an event "monitor reset init" that sets the MCU to 64 mhz (for the stm32f4x), and the adapter speed to 4mhz (instead of the normal 2mhz auto negotiated to 1800khz). This speeds up debugging / flashing etc.

Would it be possible to have the command for reset configurable?
(I tried using the `debugger_args` but that will not cut it. instead it kind of lost the notion of current executable, so i guess it replaces the "file" you give gdb on startup.)

1.5 - maybe each command sent from vscode could be configurable, but then maybe a more generic conf file would be better than putting more options in the launch config. so "pause", "step" etc. could be configurable, (not sure if all makes sense, but possibly ... I don't know.

2 -  a better gdb prompt with tabbing. with this at hand there would be no need for running gdb from a terminal. (I guess this is harder to implement...)

Besides that I have not tried out all the goodies yet, but thanks again for the excellent work!!!

Best
  Per

Marus, 2018-02-12:
Regarding your questions/suggestions 

0 - nothing speecifically implemented for semihosting - considering it - but haven’t got to implementing anything yet. 
0.5 - there is a postLaunchCommands option you can set in launch.json - that takes an array of GDB commands to run after launch/attach. (There is also preLaunchCommands which get put a bit earlier in the launch/attach process - right after connecting to the target)
1 - might consider something there so that there are points to add things in to customize reset like launch/attach
1.5 - not sure that really make sense - as those are all specific actions - but customizing the launch sequence is already in place, and reset may make sense. 
2 - have to do some research it see something like that is feasible.

perlindgren, 2018-02-13:
The "postLaunchCommands" did the trick, and it works fine.
Another minor thing, maybe adding an option to wether break at main (or not) when starting would be a good. In RTFM-Rust it will break in the reset interrupt handler (which you normally would not want). (You could set a once break in the postLaunch if you really want to break on startup (and there give  "tb <binary>::init" if you want it to break at the user entry point.

/Per

Marus, 2018-02-13:
At the moment it doesn't explicitly break at the reset interrupt handler - it simply does not start execution of the core after loading the executable. I personally preferred this to implicitly setting a breakpoint for the user and continuing to that set breakpoint (especially with the number of supported breakpoints being so limited in many cases).

perlindgren, 2018-02-22:
I'm experiencing a problem regarding restart,  

Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target
and then try again.

This happens independent if the MCU has been halted by the pause button, or a `interrupt` at the gdb prompt. I can restart by Shift-f5 and then f5. I tried restart both by the button and the Ctrl-Shift-f5.

The strange thing is that I believe I had it working before. What am I missing?

(I have checked that i don't have any .gdbinit or so laying around, messing things up.)

Best
  Per

perlindgren, 2018-02-22:
About the startup behavior, you do a `reset halt` `si` (after the `load`) or? (I did not check in your source code..

Regarding breakpoints, if you want you can set a temp breakpoint, I guess it wont consume one of 6 available. (I think your solution is good, if one wants some specific startup behavior, one can always put it in the  PostLaunch section...)

Best
Per

perlindgren, 2018-02-27:
As a side note, putting a continue in the PostLaunch does indeed continue execution, but won't inform the "cortex debug" that its running, so the GUI and actual state of the MCU won't match. You can pause execution from the gdb prompt, but still not ideal. An better option would be a flag in the `launch` script.

Marus, 2018-03-01:
Yes - at the moment it won't receive notifications if you do something that
changes the running state in the postLaunchCommands - so the VS Code UI
will get out of sync with the experience. I'm working on trying to add an
option to start and run to main in the configuration - but haven't had time
to sort it out - again because of those UI/GDB state synchronization issues.

perlindgren, 2018-03-01:
Hi

Well, if you get it to just continue on start it be fine, and later break on a breakpoint set
in the post launch, (actually I did not try that… but I guess you will still have a notification issue….).

As i mentioned I have some issue with not being able to restart without either doing a manual monitor reset init
or closing the session and re-running it from scratch (either through the GUI, red button and than play), Shift-f5, f5).
Ctrl-F5 does not work.

Any idea what could be the reason to this (i’m on the dev version of openoncd and latest release of gdb, running arch).

Regarding vscode I use the Rust RLS plugin. (Seems not to have happened much on that as of late… it works but there could be improvements, not sure if anyone is working on that…)

Marus, 2018-03-01:
Can you post your complete launch.json that you are using. On OpenOCD I'm not having any issues with restart on my setups (OSX and Ubuntu 17.10, both with openocd 0.10.0), but that is of course with my particular launch.json (both my real C projects, and a very-very simplistic rust one).

timboldt, 2019-05-19:
I've been using this plug-in with Rust, using a J-Link Edu, and it works really well! However, my debug binary got too big, so I switched to release mode with debug symbols enabled, and now it loads the binary but breakpoints don't work. Any suggestions?

Otherwise, my main wish would be for better documentation. In fact, if you want my help, ping me: I would be happy to do some editing or proof-reading.

Marus, 2019-05-20:
Yeah - you can't do a pure "release" build and expect debugging to work, as you won't have proper debugging symbols. Your workaround of a debug build with optimization level of 1 tends to generally work alright for debugging - but just be aware that you can sometimes get somewhat strange results as the optimizations that are done can remove mess with how things work.

As for documentation - yes there are currently a lot of features in the extension that could deal with better documentation. Hopefully with enough time can write some documentation and examples for the various features - but as can be expected - writing documentation can be a slow, painstaking process.

Once I get a bit more time I'm planning on starting to build out the wiki section - where hopefully people can help contribute documentation and tips that they've found helpful.

dzarda, 2020-01-22:
Inserting Rust breakpoints fails with a popup, is that expected at this point? :)
```
No line 39 in file "c:\Users\Dzarda\Documents\src\lol\src\main.rs". (from break-insert "c:\\Users\\Dzarda\\Documents\\src\\lol\\src\\main.rs:39")
```

PhilippHaefele, 2023-01-26:
It seems that nowadays Rust is working quite fine in Coretx-Debug when debugInfo ist set to the right value (see https://github.com/Marus/cortex-debug/issues/727)

@haneefdm As of this we can close this issue and if something is not working, the users should open a new issue with their exact problem

haneefdm, 2023-01-26:
Yes, I used Rust myself on an ST device. Seems to work

closed 2023-01-26 by haneefdm