Marus/cortex-debug #13: Native debug configuration

ivankravets, 2018-01-31:
How is difficult to add native debug configurations which we had in the original extension? 

See our [existing configuration](https://github.com/platformio/platformio-core/blob/develop/platformio/ide/tpls/vscode/.vscode/launch.json.tpl) for native-debug. Os, having this configuration we could switch to your extension. We don't need to specify paths to openOCD, J-LInk, or etc, because we handle this work on our virtual GDB side.

Thanks!

Marus, 2018-01-31:
In theory it shouldn't be too hard to add support for a GDB server externally launched by PlatformIO (as that would be similar to how my support for the Black Magic Probe works; as that is an always running GDB server).

There are a few things that are some restrictions/problematic points with this as I'm seeing it right now though:

1. PlatformIO would need to provide my extension with paths to its installed toolchain, as the extension makes use of more tools than just GDB and the GDB server (For example I use the arm-none-eabi-objdump utility to read a symbol table in order to be able to determine global and static variables, as well as function locations for a disassembly view; and I might make use of other tools in the future). This is now done through a user or workspace setting (cortex-debug.armToolchainPath  - which your tools can likely set as it is already generating a settings.json file for workspace settings)
2. I would suggest remaining with a different extension for your non Cortex-M platforms (While certain parts of it may work, I have currently made the assumption of ARM Cortex-M in my development as those are the only platforms I am currently working with - not to say there is 0 chance of me ever extending or doing something for other platforms in the future). There would certainly be a few issues with non Cortex-M platforms at the moment - the SVD based peripheral register viewer and reading the symbol table being the obvious ones.
3. PlatformIO would need a way to provide all the appropriate GDB commands for launching/attaching to the debugger; I'm expect this is already done satisfactorily in your .pioinit file - but since I can't locate that file when I just tried a sample project (created/deleted on the fly I'm guessing) I don't know exactly what it contains.
4. I wouldn't want to use your existing configuration as is though though, as that uses the debug type "gdb". My extension trying to use that would create a conflict with the existing native-debug extension, causing issues if anyone wants to have both extensions installed (as might be desired even for PlatformIO if you keep native-debug for your other non Cortex-M platforms).
5. In order for the peripheral register viewer to work I need to be provided with an SVD file for the target device. In a small number of cases I automatically map the device name (provided as a launch.json parameter; currently just a small subset of STM32 - I'm still trying to determine exactly what I want to do here to expand this functionality); otherwise the path to the SVD file needs to be provided as a launch.json parameter.
6. SWO/ITM output, if desired , would only be able to be collected through an external serial device (can in theory be one coupled with the probe if your probe has that and is not being used) - as collection directly through the probe's debug interface requires specific setup and interaction with the GDB server/probe (and isn't even available in all cases anyway).

Marus, 2018-02-01:
With a little bit of tweaks on my side (these haven't been released yet) - I was able to get the PlatformIO debug system working with a launch.json entry like the following (as I said - wouldn't make it work with your exact existing configuration):

https://pastebin.com/C0eR9kyx

I expect it would be fairly easy to deal with most of those configuration changes with an update to your extension (as it regenerates the launch.json every time it opens the project by the looks of it). Being able to provide an SVD file to us would of course take more work on your side - as you'd need to build a library of SVD files for each supported device. You can also supply the device name in a device parameter - and hope that I have a mapping to an SVD file available - but as I said - that is fairly limited currently and will likely take some time before I have had the time to build up a significant number of SVD files.

There was some strangeness with the behaviour of stepping / breakpoints - but I don't think that has anything to do with my extension as it seemed similar with native-debug. I think that's likely a result of your build settings (I'm guessing when you are launching for debug it is building with optimizations enabled which can always cause problems with stepping, etc).

ivankravets, 2018-04-12:
> With a little bit of tweaks on my side (these haven't been released yet)

Could release these changes? I would be glad to test it

> I expect it would be fairly easy to deal with most of those configuration changes with an update to your extension

I can provide a path to SVD file depending on the board type. I like your configuration above. I see that we should not do any changes on our side, just provide SVD file.

-------

How about to rename this extension to `embedded-debug` and make it compatible with other MCUs? If SVD file is not provided, just keep the same behavior as  `native-debug` has. I would be glad to contribute to this extension.

Thanks!

ivankravets, 2018-04-13:
Thanks! Would be thankful for the configuration for external "debugger", such as PIO. I'll try it and check how is difficult to support multiple architectures.

ivankravets, 2018-04-24:
> I would suggest remaining with a different extension for your non Cortex-M platforms

Thanks! We have started a work on the own cross-MCUs extension based on https://github.com/WebFreak001/code-debug . Also, applied your patches. Out aim is to support all hardware from @PlatformIO but not only ARM.

If we will see common issues between this extension and our, we provide a PR to this repository.

Thank you so much for your efforts!

closed 2018-04-24 by ivankravets