Marus/cortex-debug #15: Support for various number formats

johnjiabinzhang, 2018-02-05:
Sometimes it's helpful to be able to view the register values in decimal or well formatted binary. This could be useful for our teaching purpose. 

One possibility is adding options in the right-click drop down menu to change the numeric format of a particular register.

Would you be considering adding such support? Or would you like a pull request for it?

Marus, 2018-02-05:
Could certainly do that. At the moment it is all hex, with the exception of fields in registers that are 3 bits or less, but would be good to be able to set the format.

I should probably be able to do this for the next update (likely later this week).

Marus, 2018-02-07:
Just to give you an update on this - I have this implemented for the peripheral registers view now - will go out in the next release. Allows the user to select between the current automatic formatting and selecting explicitly between hex, decimal, and binary.

The settings get preserved from debug-session to debug session - and are based upon a particular sub-tree of the register view (so you can view one peripheral, or register within it, in a particular format - without it applying to others.

I'm going to try to get a chance to do the same thing to the core registers view as well before the next release.

johnjiabinzhang, 2018-02-07:
Great! 👍

closed 2018-02-10 by Marus