Marus/cortex-debug #17: Support J-Link target interface selection

weynechen, 2018-02-07:
It's very excited to found this extension, I'am a Chinese developer , and I'am using VSCode to develop STM32 program everyday , edit and debug(use cpp extension +J-Link GDB server ). 

I tried cortex-debug , but the first problem is , I use SWD interface , not JTAG , to connect to STM32 . And cortex-debug do not work for this. Can you add a configuration for interface chosen?

The command introduction at the 83 page of <UM08001_JLink>.

Anyway , it's very appreciate , and I will study your source code , to figure out how this work .

Samonitari, 2018-02-07:
Sorry for chiming in.

You can use OpenOCD (till interface parameter will be handled) if you want, as it supports the J-Link interface. You can easily find OpenOCD config files for STM32-s on the net.
I suppose the reason I can use SWD with JLinkGDBServer is because LPC82X series controllers only provide this single debug interface (and JTAG for boundary checking only) because of low their pin count, so the device parameter already sets it correctly. Propietary J-Link magic.

Marus, 2018-02-08:
Can you also send me the output that is collected in the "Adapter Output" view of the Output tab.

weynechen, 2018-02-09:
I found the reason . My OS is Windows , when JLink installed , the name of GDB server command line is `JLinkGDBServerCL.exe`, not `JLinkGDBServer.exe` . So when I use default configuration as below:
```json
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            "executable": "E:/JIG/endproduct/source/build/app.elf",
            "request": "launch",
            "type": "cortex-debug",
            "device": "STM32F103RC",            
            "servertype": "jlink"
        },
    ]
}
```
debugger call the wrong program .
```ts
	public serverExecutable() {
		if (this.args.serverpath) { return this.args.serverpath; }
		else {
			return os.platform() == 'win32' ? 'JLinkGDBServer.exe' : 'JLinkGDBServer';
		}
	}
```

Then I modified source code ,change JLinkGDBServer.exe to JLinkGDBSeverCL.exe , it is not work either.
VSCode show message `Failed to launch undefined GDB Server: Timeout.` . I tried server in command line , work normal.
```
> JLinkGDBServerCL.exe -device STM32F103RC -if SWD
SEGGER J-Link GDB Server V6.22g Command Line Version

JLinkARM.dll V6.22g (DLL compiled Jan 17 2018 16:39:42)

Command line: -device STM32F103RC -if SWD
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32F103RC
Target interface:              SWD
Target interface speed:        1000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V9 compiled Jan 11 2018 11:06:30
Hardware: V9.40
S/N: 59425868
Feature(s): RDI, GDB, FlashDL, FlashBP, JFlash, RDDI
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 2331
Connecting to target...Connected to target
Waiting for GDB connection...Restoring target state and closing J-Link connection...
```

Marus, 2018-02-09:
You are correct - on Windows it is defaulting to the wrong GDB server command - I will fix that for the next release. When I refactored some of the code to make managing the growing number of GDB servers that are supported more manageable I forgot that there are two variants of the GDB server on Windows an accidentally used the incorrect one. (And my Windows testing environment had the full path specified in a user setting, so missed it during my quick Windows tests)

In the meantime you should be able to properly set the correct command though the "cortex-debug.JLinkGDBServerPath" user setting. I expect this didn't work when you tried to edit the extension - as it would need to recompile the original TypeScript sources before it would get the changes. For example, on my Windows box that setting is the following:

"cortex-debug.JLinkGDBServerPath": "C:\\Program Files (x86)\\SEGGER\\JLink_V622d\\JLinkGDBServer.exe"

With that set this is the command that it should be using with your launch.json:

C:\\Program Files (x86)\\SEGGER\\JLink_V622d\\JLinkGDBServer.exe -if swd -port 50000 -swoport 50001 -telnet port 50002 -device STM32F103RC

(port numbers may vary, as it will look for free ports on launch - but those are the most likely ones); I'm pretty sure once you set that user setting it will function properly, please let me know if it resolves your problems. Sorry for the inconvenience.

Marus, 2018-02-11:
Hi @weynechen - just wondering if these instructions, or the new release, has fixed your issues?

Marus, 2018-02-13:
I'm going to close this issue now - if you are still having issues please get back in touch.

closed 2018-02-13 by Marus