Marus/cortex-debug #19: bmp enhancements; configurable command lines

mubes, 2018-02-16:
First of all - thanks for this, exactly what we need, a competent debug tool on a reasonably lightweight platform.  I will file a number of issues separately so they can be addressed/reconciled/rejected independently as appropriate.

The first issue is that the command lines available to the bmp configuration (and perhaps to others, I don't know) appear to be set in program code (in bmp.ts as far as I can see). That means that some assumptions and behaviours are hard-coded into the program. In my case, the use of SWD with the `monitor swdp_scan` is not appropriate, but I can also see things like multi-drop configuration, initial run-to and ITM configuration all being different depending on specific use-cases.  

Would it be possible/is it appropriate for you to add the option to override these via the launch.json configuration?

Marus, 2018-02-16:
I am looking into a few more configuration options for the BMP (as well and the other probes as well) that would allow a bit more flexibility.

In the next release you will be able to select SWD/JTAG for BMP (and J-Link) - as well as add commands to the launch/attach and restart sequences (both the end of the sequence, and near the beginning of the sequence).

If you could give some ideas of exactly what types of things you'd like to see in the configuration that may help with me deciding on future tweaks; especially for BMP as it's not a probe I really use much (In fact I don't have a functional one at the moment - I seem to have foobar'd mine the last time I attempted a firmware update and haven't got to trying to fix it yet).

The main ITM/DWT/TPIU configuration is handled based upon the configurations you set in the swoConfig section (and it is desired that these be set by the environment, as there are certain options I cannot decode currently - so it is intentional to discourage setting those options to values that would cause output it cannot decode). The main part of ITM setup that isn't done automatically is actually enabling the SWO output pin (as that varies depending on the particular microcontroller) - that can either be done in firmware code, or via the new postAttachCommands / postLaunchCommands GDB settings that will be in the next release. When I get enough time I'm going to write a few tutorials on setting that all up

mubes, 2018-02-16:
Marcel,

To be clear, I am _so_ happy that you are doing this, so please take 
this stuff as constructive criticism to encourage you to keep going, as 
it's intended, not to throw rocks.

There is always a line to be drawn between flexibility and simplicity of 
use.  The main issue I had was lack of jtag/swd switching but there will 
be other monitor commands available in the bmp ecosystem in due course 
that will also need to be accommodated (e.g. for switching between 1, 2 
& 4 bit parallel decode)...being able to add additional commands into 
the sequences (e.g. the 'start' command to run to main) would solve 80% 
of the issues, but I can imagine occasions where you simply want 
something completely non-standard.

You'll find full itm & tpiu decoders in orbuculum in C if they're any 
use to you. Once I've got the parallel trace fpga sorted (a project 
separate to, but closely associated with, bmp) then I'll look at ETM 3.5 
as well.

While I'm on, if you hit CTRL-C in the Debugger window it has no effect 
on the execution of the problem but you seem to loose communication with 
the probe and no further commands can be sent...eventually you get some 
sort of 'unsupported' response.

Where in the world are you? Perhaps someone can get a replacement probe 
to you...but PM me if you need someone to test something for you.

Regards

DAVE

Marus, 2018-02-16:
No worries, I was certainly taking this as constructive. At this stage getting feedback about features people would like to see, and problems they are encountering, is about the most valuable thing for the project I can get.

Some of the features you are wanting (like the ability to select between JTAG/SWD) are either implemented as part of the next release (which I expect I'll be pushing out some point this weekend) or at least in my roadmap

Regarding the CTRL-C issue, I will try to look into that - do you mind creating a separate issue for me to keep track of that.

I'm in Canada - but I do expect I will be able to get to mine to work again (it's one I made from an ST-Link from a Nucleo board; don't have an official one). Was trying to modify it with a custom firmware based off your Bluepill one to get your async SWO working on it, so I could try to develop the SWO support. I expect the hardware is fine and I messed something up when flashing it and killed the boot loader - if so I should just need to hook it up to an external probe and start fresh. Just haven't had time to do that at the moment (although it would be good eventually to get an official one to be able to test against the standard one as well).