Marus/cortex-debug #21: bmp enhancements; Run to start

mubes, 2018-02-16:
It would be useful in the startup commands if you could add `start` which would place a temporary breakpoint at main and run to it, rather than at the instructions after reset...of course you might on occasion _want_ to run right from reset (refer to my previous note about configurable startup options :-) ).

DAVE

Marus, 2018-02-16:
A run to main option is something I am considering - but haven't got to implementing yet.

Marus, 2018-03-31:
The new release, v 0.1.18, which should be available shortly has a new "runToMain" configuration on on launch request types. This will create a temporary breakpoint at main and run until it hits that breakpoint. Of course you shouldn't use this option if you don't have a function named "main"

closed 2018-03-31 by Marus