Marus/cortex-debug #23: Website documentaion for other debuggers

goosenphil, 2018-02-17:
The website currently only mentions the Jlink and OpenOCD server.

I'd like to see the configuration required for the STLink and black magic probe.

Could you please try to update this?

Thanks

Marus, 2018-02-19:
I do apologize for the poor documentation that is currently available; it has really lagged behind my development of the extension, so I can certainly see how this may cause confusion.

I've started on some updates now, and this page - https://marcelball.ca/projects/cortex-debug/cortex-debug-launch-configurations/ - should cover the basic basic launch configuration for all supported GDB servers. The SWO decoding, graphing, and the section on customizing the launch, attach and reset sequences (this is generally not necessary, but may be needed or desired depending on your exact situation) will hopefully be updated/added soon.

goosenphil, 2018-02-19:
Thank you, the updated documentation makes the usage very clear.

closed 2018-03-03 by goosenphil