Marus/cortex-debug #24: GDB port conflict

johnjiabinzhang, 2018-02-27:
We have found a few cases where the default port `50000` is being used by other software. This port conflict causes GDB unable to launch.

Would you be able to add an option in the debug config that can manually select the port which the server should listen and GDB should connect to?

Marus, 2018-02-28:
Ok - I’ll try to look into that - it is supposed to be auto-detecting free ports to use (I start the search at 50000 - so that’s the typical one) - so that functionality must not be working properly. 

Did this happen on the first attempt to use it (since launching VSCode) - or only in subsequent ones. If it’s only happening in subsequent launches that may indicate that the GDB server isn’t closing properly in all cases (in addition to the issue with misidentifying ports as being free).

johnjiabinzhang, 2018-02-28:
Hmm... I wasn't sure if it was the first time. The case I dealt with was that a dictionary app was using port 50000, but somehow `st-util` was still reporting to be listening on port 50000. So when GDB was trying to connect to it, it got refused.
Maybe there was some problem with `st-util` as well? It's quite a rare case. If you can't quite fix it then maybe an manual option to specify the port might be good.

Marus, 2018-02-28:
Ok - good to know that it was something external - had seen cases before where an old GDb server didn’t quit. 

I’ll try to see if I can figure out why detection may be failing - and perhaps look into an override as an advanced setting. Likely won’t get to looking at it until the weekend.

Marus, 2018-03-03:
I've just published a new update (V0.1.16) that should fix this issue. If your machine has IPv6 enabled then the open port checks were defaulting to checking for the IPv6 port instead of the IPv4 port. So if the port was in use but only bound on IPv4 (which is quite common) it wouldn't detect that the port was in use. (If you didn't have IPv6 enabled then it would check IPv4 and function properly.)

I've changed the checks to force checking the ports on IPv4 (as the GDB servers are all using IPv4 ports) - so should now properly detect the conflicts.

If you still receive this problem after the update please re-open the issue and we'll see about a manual setting work-around - but I expect this will fix the problem.

closed 2018-03-03 by Marus