Marus/cortex-debug #28: Parsing of SVD Files

grbd, 2018-03-11:
Hi,
I recently tried using an svd file
I managed to pick up one from here https://github.com/posborne/cmsis-svd
since I'm using a LPC1769 I tried using this one https://github.com/posborne/cmsis-svd/blob/master/data/NXP/LPC176x5x_v0.2.svd

However it doesn't look like this works
in the debugger it just says "No SVD File Loaded"

In order to narrow things down a bit, I tried forcing the use of one of the svd files that comes with the extension
copy paste STM32F0xx.svd into the root directory then tried
```
"svdFile": "${workspaceFolder}/STM32F0xx.svd",
```
This actually works / is parsed, although it's the incorrect mapping for the MCU it does show entries within the debug window
This suggests there's something within LPC176x5x_v0.2.svd that the parser doesn't like

I might try removing bits of the file to see if that gets it to work

grbd, 2018-03-11:
If I remove all the peripherals except for the last one within LPC176x5x_v0.2.svd (which is GPIO) then it works
If I remove all the peripherals except for the first one within the same file (which is WDT the watchdog timer) then it fails to load / gives a "No SVD File Loaded"

Marus, 2018-03-11:
Ok. I’ll have a look at that SVD file. The SVD format is kind of strange, with numerous ways to represent the exact same thing (to the point that it even has 3 methods to represent what bits in a register belong to a particular field), so possible the LPC176x5x_v0.2.svd file is using some combination that hasn’t been covered properly. 

I’m in the process on completely revamping the SVD parsing - so I’ll ensure that file works with the new system, and if there is a quick fix I can do for the current release I’ll release a hot fix to cover that particular one.

Marus, 2018-03-11:
Well - I have determined what is causing the error parsing the SVD file. In that file - the enumeration for the CLKSEL field of the CLKSEL register has an <enumeratedValue> entry without a value (instead using an <isDefault> element under the <enumeratedValue>).

In all the SVD files I have found (including the ~550 that are part of https://github.com/posborne/cmsis-svd) there is a total of one usage of that style of enumeratedValue definition (namely the one you found in the WDT peripheral of LPC176x5x_v0.2.svd).

I've release a small workaround fix (v0.1.17 - should be available as an update shortly) that should make the file parse (although I'm not handling that one enumeratedValue at this point - to be honest I'm not 100% sure what the intended interpretation of that isDefault field is yet). 

I'll make sure it is handled in the major SVD rework I'm doing (and hopefully can determine what the correct handling of that case is).

closed 2018-03-11 by Marus

grbd, 2018-03-11:
Thanks for looking at this

Marus, 2018-03-11:
Thanks for reporting the issue - something that I probably never would have come across in my own testing since it is so rarely used in actual SVD files (as I said - I've found 1 instance of it in all the SVD files I can find).

Let me know if you run into any other issues - feedback is always great to get - especially in these early stages with the tool.

grbd, 2018-03-11:
Yep I just tried the update and it's now working great