Marus/cortex-debug #31: Debugger does not pause on breakpoint when FreeRTOS is enabled

samuelsadok, 2018-03-15:
I tried the newly added RTOS support but it does not seem to work for me as intended.

Steps to reproduce:
1. Enable cortex-debug's new FreeRTOS support by adding `"rtos": "FreeRTOS"` to launch.json
2. Set a breakpoint shortly after the call to `vTaskStartScheduler`.
3. Plug in the STLink into USB and power on the target board
4. Press the "Go" symbol to start debugging
5. The debugger will break immediately after reset as usual
6. Press continue
7. The actions of the target board indicate that the breakpoint is indeed hit, but everything in VSCode appears as if the target is still running. Clicking on the pause button has no effect.

Debug console output:
```
adapter speed: 2000 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08008e20 msp: 0x20020000
Note: automatically using hardware breakpoints for read-only addresses.
[Switching to Thread 1]
```

I'm using:
STM32F405RG
STLink V2
OpenOCD 0.10.0
VSCode 1.19.0 on Linux

launch.json:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            // For the Cortex-Debug extension
            "type": "cortex-debug",
            "servertype": "openocd",
            "request": "launch",
            "name": "Debug ODrive",
            "executable": "${workspaceRoot}/build/ODriveFirmware.elf",
            "rtos": "FreeRTOS",
            "device": "STM32F405RG",
            "configFiles": [
                "interface/stlink-v2.cfg",
                "target/stm32f4x_stlink.cfg",
            ],
            "cwd": "${workspaceRoot}"
        },
    ]
}
```

Marus, 2018-03-17:
Ok - since you are using OpenOCD for your GDB - have you included the FreeRTOS-openocd helper in your project - https://github.com/arduino/OpenOCD/blob/master/contrib/rtos-helpers/FreeRTOS-openocd.c

This is required for FreeRTOS v7.5 or above - as OpenOCDs FreeRTOS support was written based upon an old version of FreeRTOS and isn't compatible with newer versions without that helper file (There are some special instructions in the comments of that file that you need if you are using --gc-sections in your linking).

samuelsadok, 2018-03-20:
Oh I must have missed that, adding this file indeed solved the issue, thanks!

The behavior is still a bit funky though: when hitting a breakpoint VSCode tends to "focus" on the wrong thread (it says "paused on breakpoint" for a random thread that was not actually the one with the breakpoint).
This is a different issue though so I think we can close this one and if at some point I collect more systematic information about the other thing I'll open a new issue.

Marus, 2018-03-21:
Ok - thanks - when I get a chance I'll update the documentation to mention that helper file. Without it GDB will get confused - and unfortunately there isn't anything the front-end can do about that. GDB with RTOS is still a bit flakey at times.

If you find any details about the other issue let me know and I can try to look into that.

closed 2018-03-21 by Marus