Marus/cortex-debug #33: Some core Cortex registers missing?

korken89, 2018-03-21:
Hi! I was testing the debug plugin, great work!

After testing some I realized that the following registers are missing in the register view, is there a way to add them or am I doing something wrong?
Thanks!

```
PRIMASK
FAULTMASK
BASEPRI
CONTROL
```

Marus, 2018-03-21:
Which GDB server and microcontroller family are you using? I know the st-util server has some issues where it doesn't properly report all of the registers (and reports a few that don't exist or are incorrectly named).

korken89, 2018-03-22:
Hi Marus, thanks for coming back!
I was testing this on a STM32F405RG with a BMP as programmer.

Marus, 2018-03-22:
OK - I'll look into it - but I expect its another case where the GDB server (the one built into the probe in this case) doesn't report all registers. Might end up being something that will need to reported to the BMP team.

Marus, 2018-03-24:
Hi - I just took a peak with my BMP here - and as I expected those 4 registers are not reported by the BMP GDB server. Unfortunately since the processor core registers are not memory mapped I do not believe I can work around the GDB server not reporting them correctly/completely. 

If you need access to those 4 registers I would suggest filing an issue with the BMP team at https://github.com/blacksphere/blackmagic - once they have fixed it in their firmware it should appear properly in the Cortex-Debug extension.

Wish I had better news and could fix it for you quickly and easily.

closed 2018-03-24 by Marus

korken89, 2018-03-24:
Thank you for looking into this, I have already contacted them with reference for a fix for it