Marus/cortex-debug #35: Failed to launch undefined GDB Server

abeck70, 2018-03-25:
Error: spawn EACCES
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            "executable": ".......elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "STM32F722ZE",
            "interface": "swd",
        }
    ]

$ which JLinkGDBServer
/opt/local/bin/JLinkGDBServer

Hi, I attempted to us cortex-debug on OSX. I can not figure out how to debug this issue as there is no verbose output available.  I also added /opt/local/bin to the JLink path in user settings. Both approaches didn't work. My setup is working in eclipse.  Thanks.

Marus, 2018-03-26:
The JLinkGDBServerPath setting in the user setting should include the executable name - so in your case that setting should be "/opt/local/bin/JLinkGDBServer" not "/opt/local/bin".

The only path user setting that does not include the actual executable name is the armToolchainPath - as multiple executables from the toolchain are needed within the same distribution.

Marus, 2018-03-29:
I'm going to assume that setting fixed your issue - if you are still running into problems please re-open a new issue.

closed 2018-03-29 by Marus

abeck70, 2018-04-01:
Thanks Marus that was the issue.