Marus/cortex-debug #36: Feature request: run openocd from a remote pc

abelmeg, 2018-03-27:
I have been using the extension and found it to be very nice and valuable tool. Now, I wanted to run openocd from another computer  in the same network. I was able to achieve this by modifying the source-code and changing the ip address and port numbers while at the same time disabling the local openocd. This is working fine but I was wondering if this is something you can consider for future releases. This can be in the form of providing optional arguments to disable the local openocd and provide the remote address and port numbers. The remote openocd can be started manually.

Thanks

Marus, 2018-03-28:
This is certainly something I will consider for future versions - but don't know exactly when I will get to it. Thanks for the suggestion.

mariusgreuel, 2018-08-25:
With cppdbg, you would start openocd manually and adjust "miDebuggerServerAddress": "localhost:3333" to point to the remote hosts gdb server.

shlomow, 2019-01-26:
Hi Marus,
Is there any news regarding this feature?

Marus, 2019-01-27:
This is something that is still fairly low priority for me to officially support.

There is technically a hidden "external" server type which may be able to support this now - something similar to the following for your launch.json may work (The syntax highlighting may show `"server type": "external"` as an error, but you should be able to ignore it).

```
{
  "name": "Debug with External",
  "cwd": "${workspaceRoot}",
  "executable": "<path to executable>",
  "request": "launch",
  "type": "cortex-debug",
  "runToMain": true,
  "servertype": "external",
  "preLaunchCommands": [
    "target extended-remote <hostname/ip>:<port>", 
    "monitor halt"
  ]
}
```

(If you don't want to flash the remote target with the firmware; then you can use `"type": "attach"` and replace "preLaunchCommands" with "preAttachCommands"; and you may or may not need the `monitor halt` command)

In this case though it doesn't know that it is specifically connecting to OpenOCD - so some features may to work properly that would otherwise work in an OpenOCD configuration (e.g. RTOS support, SWO output)

tbandtg, 2021-01-26:
Has anyone gotten this to work? I assume you fill in your ip address as such.
            "request": "launch",
            "type": "cortex-debug",
            "runToMain": true,
            "servertype": "external",
            "preLaunchCommands": [
              "target extended-remote <hostname/ip>172.27.9.47:3334<port>", 
              "monitor halt"
            ]
            
            but I get an error, 
            
            External GDB server type must specify the GDB target. This should either be a "hostname:port" combination or a serial port.

haneefdm, 2021-01-27:
I use it all the time from more than a year ago. Here is what is in the change log
. I always use `gdbTarget` to specify host:port

* Added new `servertype` of external for cases where you want to control the GDB server yourself. This could be used for cases where you need to run the GDB server on a different machine, or in cases where there are multiple target cores which may cause the debug server to not operate as expected by cortex-debug. This configuration may require more customizations to the launch.json file than other more automated server types. When this `servertype` is selected a value for the `gdbTarget` launch.json property must be supplied.
* Added new launch.json options (overrideLaunchCommands, overrideRestartCommands and overrideAttachCommands) to be able to override the default commands run on those operations. In most cases this is not needed, but may be required for `external` server types (by default commands that are compatible with openocd are used for `external` server types).

tbandtg, 2021-01-27:
So I added teh gdbTarget
"gdbTarget": "172.27.9.47:3333",

IT says it can not connect to the gdb

Do you have an example of how you overrode the different attach and launch commands?

[image]
[image]

[image]