Marus/cortex-debug #38: Strange behaviour of this extension in my side

olivierpe, 2018-03-28:
Hello,
I try since several times to run my project with your extension, it does not run as it should but I do not know where to start to debug:

I am able to run my project using a classic cppdbg configuration (here after):
        /*{
            "name": "(gdb) Proj",
            "logging": { "trace": true, "traceResponse": true, "engineLogging": true },
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceRoot}/Proj",
            "args": [],
            "stopAtEntry": true,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": true,
            "MIMode": "gdb",
            "miDebuggerPath": "/opt/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gdb",
            "miDebuggerServerAddress": "localhost:2331",
            "debugServerPath": "/opt/SEGGER/JLink_V614f/JLinkGDBServer",
            "debugServerArgs": "-device STM32F429ZI -if SWD -speed 10000",
            "serverStarted": "connection...",
            "preLaunchTask": "flash",
            
            "setupCommands": [
                {
                    "description": "Exec file and symbols",
                    "text": "-file-exec-and-symbols ${workspaceRoot}/LpdModeImage",
                    "ignoreFailures": true
                }
            ]
        }*/

I just had to make a flash target in my makefile that directly uses Jlink to flash

ANd with the cortex M debug configuration as this:
        {
            "name": "Debug Microcontroller",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "executable": "${workspaceRoot}/Proj",
            "cwd": "${workspaceFolder}",
            "device": "STM32F429ZI",
            "interface":"swd",
            "showDevDebugOutput": true,
            "debugServer": 2331,
        }

I get the first time "connection refused": I assume this was because the Jlink gdb server path was not correct, I found in another issue these parameters in my preferences settings:
    "cortex-debug.JLinkGDBServerPath": "/opt/SEGGER/JLink_V614f/JLinkGDBServer",
    "cortex-debug.armToolchainPath": "/opt/gcc-arm-none-eabi-5_4-2016q3/bin", (I tried too with a terminating "/")

But I still have the "connect ECONNREFUSED 127.0.0.1:2331" message

If I start manually the JlinkGDBServer, I do not have this message, but the debug does not start, event if I get in the JlinkGDBServer side a connection (Waiting for GDB connection...Connected to 127.0.0.1
)

Would you have some tricks top enable debug verbosity so I can understand what happens ?

Thanks

Marus, 2018-03-28:
The debugServer is not the port for your GDB server - that is actually a port for when I am developing the extension - allows me to run the debug extension in a special mode where I can debug the debugger extensions.

You don't need to specify the port for the GDB server in the configuration at all, or launch the GDB server yourself, the extension takes care of launching the GDB server and determining available ports to use. I expect it will likely work if you remove the "debugServer": 2331 line. 

You will also likely want to remove showDevDebugOutput - again that is more intended for me debugging the extension itself than using the extension; it will print out all of the commands and responses to/from the GDB server (in GDB MI format making them hard to read).

If that doesn't fix your issues please let me know and we can investigate more.

Marus, 2018-03-28:
Oh - and assuming those paths are the correct locations on your system (and they look reasonable) they should be fine. Don't think the trailing / will matter on the armToolchainPath setting (although when I get a chance I'm going to confirm that to make 100% sure) - and the JLinkGDBServerPath is correct to include the actual executable name.

olivierpe, 2018-03-29:
Hi Marus,
it is all right, by just removing these two parameters, thanks !

Marus, 2018-03-29:
Thanks for confirming, I’ll close the ticket now and when I get a chance I’ll update the documentation to mention the debugServer command as I can see that one being confusing. If you run into other issues don’t hesitate to get back in touch.

closed 2018-03-29 by Marus