Marus/cortex-debug #40: How to set a breakpoint at main

ffzwadd, 2018-04-02:
Hi, I am trying to use your plugin, it works but I want more...

launch.json:

{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Something Launch",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "device": "STM32L433CC",
            "cwd": "${workspaceRoot}",
            "executable": "build/something.elf",
            "interface": "swd",
            "debuggerArgs": [
            ],
            "svdFile": "/Users/path to/Keil/STM32L4xx_DFP/2.0.0/CMSIS/SVD/STM32L4x3.svd"
        }
    ]
}

When press F5 the debug console shows:
/Users/someone/.gdbinit:1: Error in sourced command file:
No symbol table is loaded.  Use the "file" command.
Reading symbols from /Users/path to/project/build/LoRa_eTag.elf...
done.
0x08000df2 in main () at Core/Src/main.c:101
101	  MX_RTC_Init();
Not implemented stop reason (assuming exception): undefined
Resetting target
Resetting target

The program halts at: 

ldr   sp, =_estack

I want to set a breakpoint at main ant halt at main when we launch debug.

Marus, 2018-04-03:
By default the debug process will flash the microcontroller and reset the target core, but leave the target core in a halted state. This will leave you at the start of your reset handler (This is the first value in your interrupt vectors after the initial stack pointer). My guess would be that **ldd sp, =_stack** is the first instruction in your reset handler (this seems reasonable, as most reset handlers would setup the stack, BSS and data sections in RAM).

Optionally you can specify the "runToMain" option in the configuration - this will set a temporary breakpoint at the start of your main function and start the target core running. Which should then very shortly (assuming your reset handler is functioning properly and a function named main is your main entry point) be hit - leaving your halted at the start of your main function.

Alternatively you can set whatever breakpoints you wish and then hit the continue button in the control strip at the top of the windows.

ffzwadd, 2018-04-03:
Thanks Marcus, could you please show me how to specify the "runToMain" option in the configuration.

Best regards,
Charles

Marus, 2018-04-03:
In the appropriate launch configuration in your launch.json file you would have "runToMain": true

The runToMain option was only added in the latest release - so make sure you are on the 0.1.18 version.

ffzwadd, 2018-04-03:
Thanks for your help Marcus, it works. I think it's better to have a user guide for the beginners like me.

Best regards,
Charles

closed 2018-04-03 by ffzwadd