Marus/cortex-debug #41: Adding more tag on description.

kbumsik, 2018-04-05:
Hello,
Firstly thank you very much for this awesome extension. It does exactly what I need. I would like to share my experience and simeple suggestions here.

When tried searching extensions for embbeded system on VS code for the first time, I could not find this extension. This is because I my search words was "openocd" and "jlink" and this extension did not appear on the search result. I missed this extension for months.

So why not make this extension more searchable, e.g. adding `openocd`, `jlink`, `stlink`, and etc tags in the extension description?

Thank you.

Marus, 2018-04-05:
Thanks for the suggestion. I’ll update to have more complete tags on the next release.

Marus, 2018-04-09:
Unfortunately I cannot actually add any more. The VSCode extension manifest limits me to 5 explicit keywords - so I can't add one for each of the GDB servers used.

closed 2018-04-09 by Marus