Marus/cortex-debug #44: JLink and SVD files

qavaxb, 2018-04-06:
I started to use this plugin with JLink.
From configuration point of view _device_ setting is overriding _svdFile_.
The problem is, that I'm using Silicon Laboratories microcontrollers and some of symbols in device type are _XXX_ for JLink _device_ setting. SVD files are available for exact model of chip.

Would it be possible to change behavior of _device_ and _svdFile_ dependence?
Or might it be feasible to supply SVD files for other vendors?
On the other hand many SVD files will increase size of plugin...

qavaxb, 2018-04-09:
I am aware, that JLink requires _device_ option to be set.
I would rather loose dependency between _svdFile_ and _device_ setting. If _svdFile_ is not being implicitly set, then it would try to get information from _device_ setting, otherwise _device_ and _svdFile_ might be independently interpreted. This would simplify handling of SVD files and remove necessity to provide them bundled into plugin.

Marus, 2018-04-09:
I'm still not understanding what your issue is though. If you want to specify your own SVD you can do so through the *svdFile* launch configuration option - that should always take priority over what it may auto-determine based upon the *device* option - so you still have full flexibility to specify your own file outside of the ones that I do bundle.

The use of the *device* launch configuration option to determine a bundled SVD file is simply a convenience for people who happen to use one of the devices that are bundled so they don't have to locate and configure their own SVD file.

Marus, 2018-04-24:
Since you haven't given any additional feedback and I'm unsure of what the problem you are encountering is I'm going to close the issue. If you have additional feedback please re-open the issue.

closed 2018-04-24 by Marus