Marus/cortex-debug #46: VSIX install fails for Visual Studio Professional 2017

apena1970, 2018-04-17:
I have Visual Studio Professional 2017 and the VSIX installer fails. 
Log shows:

VSIXInstaller.NoApplicableSKUsException: This extension is not installable on any currently installed products.

The complete log for the installer is also included.
Is this an issue with the installer or no support for this version of Visual Studio?

[VSIXInstaller_743f9e52-273e-4354-a212-7ad2c26bd984.log](https://github.com/Marus/cortex-debug/files/1921634/VSIXInstaller_743f9e52-273e-4354-a212-7ad2c26bd984.log)

Samonitari, 2018-04-18:
It is a plugin for VS Code, which is a completely different application that the old, Visual Studio

Marus, 2018-04-18:
As @Samonitari pointed out - this is not an extension for Visual Studio - but an extension for Visual Studio Code.

closed 2018-04-18 by Marus