Marus/cortex-debug #47: ITM debugging doesn't work under Windows

bergus, 2018-04-24:
When I try to set `swoConfig.enabled` to `true` in my launch.json configuration entry, I get an exception:

> Failed to launch OpenOCD GDB Server: Timeout.

In the adapter output, it starts almost normally, but then it logs

> Error: Can't open trace destination file

which appears to come from [here](http://repo.or.cz/openocd.git/blob/HEAD:/src/target/armv7m_trace.c#l187), bitching about the [`tpio config internal` command](http://openocd.org/doc/html/Architecture-and-Core-Commands.html#ARMv7-and-ARMv8-Architecture) being called with an invalid path - [here in the extenion](https://github.com/Marus/cortex-debug/blob/86bd82dac4c7013e7ad6278ba6fa980d153c1813/src/openocd.ts#L136-L139), I gather. I also can see that `ChildProcess.spawnSync('mkfifo', [this.swoPath])` is only called on non-windows platforms, so this is probably the culprit - the file really doesn't exist.

**What do you think can I do about this? Do you want to support this at all?**

----

[This SO thread](https://stackoverflow.com/q/11750041/1048572) suggests using sockets instead. I have no idea what openocd expects to work with.
I also have `mkfifo` tools installed from Mingw64 and Cygwin64, could they work if I just remove the OS check?

Marus, 2018-04-24:
No, it is intentional that mkfifo is not called on windows, as Windows does
not support Unix style fifos that are used on OS X and Linux. More likely
it is not determining a temporary file path correctly (or for some reason
you do not have permission to write to the temporary path it determines).

What versions of Windows, VS Code and OpenOCD are you using?

Marus, 2018-04-25:
> Win7, VS Code 1.22.2, OpenOCD 0.10.0+dev-00404-g20463c28
>
> I've also tried running VS Code with admin permissions, didn't change
> anything. I guess I'll try to log the temp file path.

Ok, this weekend I'll try to setup a similar environment to test and see if
I can find anything that may be an issue. My windows testing environment is
Win 10 - so may be something about the temporary path detection creation
that is only working properly on Window 10 (could also be some other
permissions issue local to your machine).

bergus, 2018-04-26:
OK, I think I solved this.
(I couldn't get any kind of [debugging for the debug adapter](https://code.visualstudio.com/docs/extensions/example-debuggers) to work, so I had to make do with `fs.appendFile`. Ouch. That's worse than `console.log`-debugging.)

The `commands` were something like
```
tpiu config internal C:\Users\…\AppData\Local\Temp\tmp-59205ueCOKVRfBUg uart off 0 0-f
```
and I could reproduce the issue in the native `gdb` console - it just couldn't open the file. Even if the file was previously created. So my default suspicion #&#xfeff;1 with command line args: blanks are not escaped in the file path. True, but there are no blanks in the path. Suspicion #&#xfeff;2 with windows file paths: `\` needs to be replaced by `/`. Um, tried
```
tpiu config internal C:/Users/…/AppData/Local/Temp/tmp-59205ueCOKVRfBUg uart off 0 0-f
```
and **it works**! So a simple
```js
if (os.platform() == 'win32') {
   this.swoPath = this.swoPath.replace(/\\/g, "/");
}
```
helps.

bergus, 2018-04-26:
Next problem (not sure whether this should be a separate issue to report) is that I am getting

> Failed to launch GDB: No symbol "NaN" in current context. (from interpreter-exec console "BaseSWOSetup NaN")

Looking through the code, this appears to come from `SWOConfigurationCommands()` when there are no frequencies specified - despite the help message saying "*0 will attempt to automatically calculate*". (Setting only the CPU frequency leads to the same problem with `Infinity` instead of `NaN`). Unfortunately I could not find any documentation about these commands on the web, neither in the OpenOCD docs nor in the GDB docs nor with a search engine, so I don't know how to fix that. [This article](http://blog.japaric.io/itm/#itm) says that the second number isn't required at all, at least in the `tpiu` command.
I guess I'll have to investigate further to get this running.

bergus, 2018-04-26:
Oh well, I just figured out with help of [this thread](https://developer-sjc-indigo-border.mbed.org/questions/80963/SWO-on-nucleo-L432KC/) that the SWO pin isn't available on my Nucleo STM32L432KC board anyway… (And I'm not yet ready to try manually connecting it).
Still I'm happy to run your debugging extension without any errors, and I got semihosting working so that will need to be sufficient for now.

Marus, 2018-04-26:
Thanks for all the feedback. File paths on windows can sometimes be strange - I'll get the fix in there once I've confirmed that it doesn't cause any problems on Win 10 (at the SWO/ITM did work for me as it currently is on Windows, so could be a difference between OpenOCD and or Windows versions). 

I will update the documentation about the SWO speed; the 0 option for automatic is left over from when I only supported J-Link as a debug probe, never got updated to reflect the differences (or apparently validate the input) when I added OpenOCD support for SWO.

Strange that they didn't connect the SWO output pin on the small Nucleo-32 Boards to the onboard ST-Link, as it is connected on the Nucleo-64 and Nucleo-144 boards; good to know in case other people end up having the same question.

If you do run into other issues please let me know.

nergnezor, 2018-05-04:
The proposed fix works for me too.
I'm using Win 10 with OpenOCD 0.10.0 and VS Code 1.23.0

Marus, 2018-05-05:
The fixed path generation, and removing the mention of auto-calculating frequencies, will go out with the next update.

closed 2018-05-05 by Marus

klaus-liebler, 2020-02-27:
Hi Marus,
there is still an issue with this in v0.3.4 with VS Code 1.42.1 in Win10.
Due to whatever reason, my system has two user environment variables TMP and TEMP pointing to directories below my user directory. Unfortunately, my user name has a space in it. When I start debugging, I get the "Error: Can't open trace destination file".
I removed the user environment variables so the system wide TMP and TEMP variables get valid (both pointing to c:\windows\temp - without spaces:-) ). Now it works perfectly...

Regards,
Klaus

haneefdm, 2020-02-27:
This is a problem I think is in an external package `tmp` we use to create temporary file names

I submitted a bug over there https://github.com/raszi/node-tmp/issues/246

Not sure if we should apply a workaround or wait for a fix.