Marus/cortex-debug #49: Failed to launch undefined GDB Server: Timeout

Spritkopf, 2018-04-30:
Hi,
I have no success starting a JLINK debug session on Windows. I am using the following launch configuration:
```
{
    "version": "0.2.0",
    "configurations": [
        
        {
            "name": "Debug (J-Link)",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "cwd": "${workspaceRoot}",
            "executable": "./build/temp-control-f4.elf",
            "device": "STM32F407VG",
            "interface": "swd",
        }
    ]
}
```

The JLinkGDBServer launches successfully and waits for a connection:
[image]

But after several seconds, I get an error message which says: **Failed to launch undefined GDB Server: Timeout**.

I am assuming, somethin with the path to the actual arm-none-eabi-gdb executable is not right? I have these lines in my workspace settings:
```
"cortex-debug.armToolchainPath": "C:\\Program Files (x86)\\GNU Tools ARM Embedded\\6 2017-q2-update\\bin",
"cortex-debug.JLinkGDBServerPath": "C:\\Program Files (x86)\\SEGGER\\JLink_V620f\\JLinkGDBServer.exe"
```

`armToolchainPath` is the one with the arm-none-eabi-* executables.

Is there anything else I forgot to configure?

Thanks!

Marus, 2018-05-01:
The issue is your cortex-debug.JLinkGDBServerPath setting - that setting needs to point to the command line version of the GDB server - not the GUI version (Should have the executable name JLinkGDBServerCL.exe on windows)

Spritkopf, 2018-05-02:
It was indeed the setting of JLinkGDBServerPath. With your suggestion it works now. 
Thanks!

unkls, 2019-05-14:
Hello,
I'm facing the same issue, but I'm on linux. JLinkGDBServer is in my $PATH but he doesn't find it. I also try like you said in your comment with the direct path but the error is still there.

`{
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}/project/make",
            "executable": "./bin/project.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "STM32L4R5",
            "interface": "swd"
 }`

Thank's in advance for your help.