Marus/cortex-debug #51: Move cortex-debug.armToolchainPath to launch.json

microwavesafe, 2018-05-01:
I have several different versions of toolchains installed, so having a single setting in settings.json is slightly awkward to use.

It would be a big help if I could set the path in launch.json
```
"configurations": [
        {
            "armToolchainPath": "/path/to/toolchain/",
```
would this be a possible?

microwavesafe, 2018-05-02:
With existing projects I do not change the toolchain ever, as this can introduce bugs. But I do update the toolchain for new projects. I was imaging having two open in a workspace built against different versions of gcc.

Thinking about this though, as it is only the gdb part of the toolchain, it may not matter that I'm not using the exact same version?

Marus, 2018-05-02:
It currently uses two tools in the toolchain, GDB as you noted, and objdump which is used to get a symbol table from the binary. Unlikely that either of those would have problems (at least for minor version bumps).

And while I think most people would use a separate workspace for each project, multiple related projects (which may still be using different toolchain versions) could reasonably be in the same workspace. I'll look at adding an override for that in the launch.json

Marus, 2018-05-22:
This was added in the latest release. The user/workspace settings still exist - but the launch.json setting should take priority.

closed 2018-05-22 by Marus