Marus/cortex-debug #53: No SVD file loaded issue

pga23, 2018-05-01:
Debugger works fine except loading SVD file.

My config is:

``` json
{
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "./build/firmware_flash.elf",
            "name": "Debug",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "configFiles": [
                "/usr/share/openocd/scripts/board/atmel_saml21_xplained_pro.cfg"
            ],
            "svdFile": "${workspaceRoot}/doc/saml21b/svd/ATSAML21J18B.svd"
        }
    ]
}
```

The official SVD file comes form [http://packs.download.atmel.com/](url) website.

I use Ubuntu 14.04, Visual Studio Code 1.22.2 and OpenOCD 0.10.0

Marus, 2018-05-02:
I'll take a look into it and see if I can determine why it may not be parsing the SVD properly.

ivankravets, 2018-05-02:
See https://github.com/Marus/cortex-debug/pull/50

Marus, 2018-05-22:
I think this should be fixed in the latest release. If the problem still persists please re-open the issue.

closed 2018-05-22 by Marus

adammunich, 2019-01-01:
I'm not so sure the problem is fixed, the same "not-loaded" bug seems to appear with this file: https://raw.githubusercontent.com/posborne/cmsis-svd/master/data/STMicro/STM32F405.svd