Marus/cortex-debug #57: Add to watch fills in wrong variable name

microwavesafe, 2018-05-02:
When right clicking on a local variable, and selecting "Add to watch", it adds "var_variable" to the watch window when it should be "variable".
When adding an array element it adds "var_array.0" for element 0. It should be "array[0]".

You can manually edit this to work, but it would be nicer if it filled it in correctly to being with.

P.S. I hope you don't mind all the issues I'm creating, I just really like the tool and want to help make it better.

Marus, 2018-05-02:
Will look into it; and don't worry about bug reports - the more issues people find the more I can fix and improve the tool for everyone.

microwavesafe, 2018-05-22:
New version has been working so far in my brief testing

Marus, 2018-05-22:
Thanks for letting me know.

closed 2018-05-22 by Marus