Marus/cortex-debug #58: Feature to view variables as different types

microwavesafe, 2018-05-03:
Would it be possible to add the ability to view the variables as different types? Working with hardware means viewing the variables as hex and even binary can be really useful. Pointers are converted to hex and ASCII is shown next to bytes, so I guess it's possible?

Also array pointers only show up as a single value, again it would be very useful if you can choose to display these as an array. Usually you are asked to enter the size of the array, then it just displays that area of memory as the variable.

Marus, 2018-05-05:
The peripheral and core register views already have the ability to set formatting (right click on the node and you will get an option to change the format).

Unfortunately the presentation of the variables in the main variable view are controlled by VS Code itself, so I have limited ability to control how it is presented, I only supply the raw data to VS Code through an existing interface, it decides how to present those. I may be able to make some tweaks but I'll need to research.

As for arrays, if the variable is a statically sized array (e.g. if it was declared as int[10]) it should already allow you to see each element. For pointer based ones it won't know the size, and again as a result of not controlling the presentation I cannot currently change the presentation to my knowledge. 

I will do some research into this area of the VS Code API to see if there is anything I can take advantage of - but may not currently be possible for me to do this.

ivankravets, 2018-05-05:
@Marus it seems that VSCode has API for that:
```
interface Capabilities {
....
        /** The debug adapter supports a 'format' attribute on the stackTraceRequest, variablesRequest, and evaluateRequest. */
        supportsValueFormattingOptions?: boolean;
....
}
```

microwavesafe, 2018-05-05:
I was looking to see if there were any existing issues for VSCode this one suggests it can be accomplished in the debugger https://github.com/Microsoft/vscode/issues/28025.

Interestingly the last comment suggests you can use "variable,h" to display the variable as hex in a watch window. Which could be a workaround for now.

In eclipse you enter the length of the array to display, maybe we could open another tab like the current memory view does? Although having it in the variable list would be preferable.

Marus, 2018-05-05:
I'll take a look at the interface that Ivan pointed out; that may give us what we are looking for, at least in the watch view. Have to see exactly what the capabilities are as I hadn't come across that on before.

I don't really like the idea of opening a new tab like the memory dump does for this, as it's confusing to have variable views presented in different places. Will take a look at what is available through the API Ivan pointed out, and perhaps make requests to the VS Code team for where I would like to see expansions to the VS Code APIs for what isn't currently possible.

RolfNoot, 2019-01-14:
Another option is to set the GDB radix using the debug-console:
set output-radix 16

Now all watch windows are in hexadecimal.

stephendpmurphy, 2019-10-02:
So is it ever planned to add this option natively to Cortex-Debug?

haneefdm, 2019-10-02:
I don't think VSCode actually supports `supportsValueFormattingOptions`. It is in the protocol but there is no actual support for it in VSCode proper. Near as I can tell.

Without help from VSCode, I don't think this is possible. @Marus did add formatting options for watch expressions in the form of `expr,format` where expr is any valid C expression and format is one of 

```javascript
        b: 'binary',
        d: 'decimal',
        h: 'hexadecimal',
        o: 'octal',
        n: 'natural',
        x: 'hexadecimal'
```

haneefdm, 2019-10-02:
Yup, no support for it in VSCode. Got confirmation. See link below

https://github.com/microsoft/VSDebugAdapterHost/issues/10

closed 2019-10-02 by microwavesafe