Marus/cortex-debug #59: fails to find the source code

kunugiza, 2018-05-04:
Debug session fails to find the corresponding source code. The underlying environment (gdb and OpenOCD) seems to be working good. I am able to download the code to the target, start a debug session, and halt the target. However, on halt, I always get disassembly rather then the source code. Addition to this, breakpoints set on source code lines are also not working.

The target binary is compiled with dwarf debugging symbols. The gdb running in the debug console itself is seeing the symbols like below. Breakpoints set manually through the console also works as expected.
```
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x08001b5c msp: 0x20001800
Note: automatically using hardware breakpoints for read-only addresses.
Temporary breakpoint 1, main () at Src/main.c:83
83	  HAL_Init();
```
I could not figure out how to make the VS Code be aware of the source code. Is there a way to forcibly tell the existence of the source tree? 

VS Code in question is running under Windows 10. I am using gnu-mcu-eclipse-openocd-0.10.0-7-20180123-1217-win64 and gcc-arm-none-eabi-7-2017-q4-major-win32-sha2 for debugging.

Marus, 2018-05-05:
Have you relocated the source tree since it was compiled? The paths in the debug symbols may not match up in that case. This can also happen with certain build setups as well - you may need to use -fdebug-prefix-map to remap the debug symbols to the correct paths. 

Can you add the "showDevDebugOutput": true to your launch.json configuration and then send me the output you get in VS Code in the main debugger output, and the Adapter Output tab (perhaps post on something like pastebin as it will be quite long) as well as your launch.json configuration. May help me determine exactly what is happening.

kunugiza, 2018-05-05:
Thank you Marus, From your hints, I finally got this solved. The cause seems to be non-ASCII characters (which was Japanese) in the workspace folder path name. After I have renamed the folder so that the full path name consist only of ASCII characters, and recompiled the project, everything has started to work well.

I'm not sure of the root cause, but it looks like something is not happy with non-ASCII characters in the path names.

Marus, 2018-05-06:
Interesting, I'll try to take a look into that if I get a chance. Could be
something on my side that isn't interpreting the non-ASCII characters in
paths properly, or could be something either in VS Code itself or GDB. Will
let you know if I find anything, but at least you have a workaround for the
time being.

thrasher8390, 2019-01-31:
@Marus, can you post an example of the launch.json with the use of -fdebug-prefix-map.

I have the issue where my .elf is generated on a different machine then where I'm debugging and so I need to remap the source files so the VS Code can find them.

Marus, 2019-06-09:
You should be able to pass additional parameters to GDB with the `debuggerArgs` parameter in launch.json.

webees, 2019-08-30:
@Marus @kunugiza 
https://wizardforcel.gitbooks.io/100-gdb-tips/directory.html

My static library can't find the source code either.

haneefdm, 2025-01-17:
You can also use pre/post launch commands to add gdb commands. Closing this for lack of activity and nothing actionable

closed 2025-01-17 by haneefdm