Marus/cortex-debug #60: SWO instructions (JLink)

0ge, 2018-05-08:
Great work with the plug-in! Works flawlessly for debugging and easy to set up for custom hardware. Kudos!

I'm trying to get SWO output from my target, using a Segger JLink. My `launch.json` looks like the following:

```
{
        "cwd": "${workspaceRoot}",
        "executable": "${workspaceRoot}/build/bin/MyApp.out",
        "name": "Debug Microcontroller",
        "request": "launch",
        "type": "cortex-debug",
        "servertype": "jlink",
        "device": "nRF52832_xxAA",
        "interface": "swd",
        "svdFile": "${workspaceRoot}/vendor/nRF5_SDK_12/svd/nrf52.svd",
        "swoConfig": {
            "source": "probe",
            "enabled": true,
            "swoFrequency": 2000000,
            "decoders": [
                {
                    "port": 0,
                    "label": "Output",
                }
            ]
        }
    }
```

From the debug terminal I see the following:

> Received monitor command: SWO EnableTarget 0 2000000 0x1 0
> SWO enabled successfully.

However, I don't get any output anywhere, and I'm not sure where to expect it.

SWO is initialised in code using the following C++ snippet. The target is a Cortex-M4 (nrf52832 from Nordic).

```
const uint32_t current_config = static_cast<uint32_t>(NRF_CLOCK->TRACECONFIG
        & ~CLOCK_TRACECONFIG_TRACEPORTSPEED_Msk);
const uint32_t speed_config =
       static_cast<uint32_t>(CLOCK_TRACECONFIG_TRACEPORTSPEED_4MHz);
NRF_CLOCK->TRACECONFIG = current_config | speed_config;
ITM->TCR |= 1U;
```

Any ideas greatly appreciated!

Marus, 2018-05-08:
OK - there a few of things that may be an issue here.

# In the decoder you should have a type specified, for text output you can use "type": "console" (you can also set an "encoding" key if you are not using plain ASCII characters in your output).
# I recommend specifying both swoFrequency and cpuFrequency in the swoConfig section. While J-Link can supposedly determine it automatically I haven't had perfect success with that myself.
# While I don't know the NRF52 series, so I can't be sure of this, it looks to me like in your snippet to initialize the SWO output in your firmware you set the trace port to 4MHz (This may be too high depending on your J-Link version/variant) - in the launch configuration you seem to be setting it to 2 MHz (which is probably more reasonable for most J-Link variants) - but whatever the actual value is they should be matching.

As for where to expect the output if it is working, that should show up as documents within the Visual Studio Code "Output" tab at the bottom - should have a label in the dropdown that includes the port number, your label, etc.

0ge, 2018-05-08:
Thanks for your reply!

- Gotcha - added that.
- Gotcha - added that too.
- Curiously enough, the datasheet for NRF52 says (regarding the trace config register) `Speed of Trace Port clock. Note that the TRACECLK pin will output this clock divided by two.`, and 4MHz is the lowest setting. I've interpreted this as the SWO frequency is 2MHz. I tried changing the launch configuration to 4MHz, but that didn't change anything.

Unfortunately, still no output, but now I know where to look - thanks! Maybe the J-Link is too slow. (Which is a bummer, since apparently the output can not be configured any slower)
Anyway, I got a feeling it's not related to the plugin, so I'll close this. Thanks for your support and hard work!

Marus, 2018-05-08:
Hum, odd that it only goes that slow for the SWO frequency. When I get a chance I'll take a peak at the NRF52 documentation and see if I can find anything else as a hint - unfortunately I don't have one of the NRF52 boards to try things myself (Have some NRF51 boards that I've used, but those are Cortex-M0 and don't have SWO).

Marus, 2018-05-08:
Check out this one - in particular where it talks about the ENABLE_SWO define - your pin might currently be operating as a GPIO not an SWO output

0ge, 2018-05-09:
[Here](http://infocenter.nordicsemi.com/pdf/nRF52832_PS_v1.4.pdf) is the datasheet. The section I was referring to is found in §19.3.10, p. 108. I'll update this issue if I find out what was the issue.

0ge, 2018-05-21:
Odd - I get output running `swoview 4000000` in `JLinkExe`, but nothing in VSC. (Even if I change the frequency to 4000000)

0ge, 2018-06-12:
Sorry, I'm going to reopen this...

 1. If I start a debug session and then quit (letting it run without the debugger), I can run `JLinkExe` and from there connect and start SWO (`swoview 4000000`).
 2. If I start a debug session manually (using `JLinkGDBServer`, `arm-none-eabi-gdb` and `telnet`) I get SWO output.
 3. If I start a debug session and try to telnet to it (port 50002) there is no output.

To get the output working with GDB I do the following (3 terminal windows required)
```
JLinkGDBServer -if swd -device nrf52832_XXAA

arm-none-eabi-gdb
(gdb) file path/to/executable.out
(gdb) target remote localhost:2331
(gdb) mon reset
(gdb) mon SWO EnableTarget 0 0 1 0
(gdb) mon swo start 64000000 2000000
(gdb) continue

telnet localhost 2333
```
Still no output in VSCode though!
I get a feeling that whether its 2000000 or 4000000Hz that should be used for SWO changes sometimes. 🤷‍♂️ 

JLink: JLink Base
Target: nRF52832
VSCode: 1.24.0
Cortex-Debug: V0.1.20
GDB Server: SEGGER J-Link GDB Server V6.32b

I've been in contact with Segger, but since it works with GDB, they consider it out-of-scope for their support.
Any ideas?

Marus, 2018-06-15:
The syntax you use there is slightly different than what I've seen documented (a separate EnableTarget and start command). 

But the first thing I would try is to specify the CPU frequency in your launch.json configuration. The one you posted would be (assuming the start command operates how I believe it would) for a cpuFrequency of 64000000 and a swoFrequncy of 2000000 - currently you're not specifying it so would perhaps not be auto-detecting it correctly.

For STM32F1 targets as well I don't actually need to configure the speed for the SWO in firmware - so you could try disabling that in your firmware as well, but would try the configuration changes first.

0ge, 2018-06-15:
Thank you for your reply.

I think Cortex-Debug issues the EnableTarget command. If I go to adapter output I can see `Received monitor command: SWO EnableTarget 64000000 2000000 0x1 0`. However, if I use the above GDB commands, but replace EnableTarget with what Cortex-Debug issues (still keeping the reset, including start command), I don't see any output. In fact, I can skip the EnableTarget, but must keep the start command. I didn't find any info on the start command in the JLink manual, but maybe in the [J-Link GDB Server - GDB protocol extensions](https://www.segger.com/downloads/jlink/#Documentation).
I have tried the following combinations of CPU frequency and SWO frequency, without any luck.

CPU freq. | SWO freq.
----------|-----------
0         | 0
0         | 2000000
0         | 4000000
64000000  | 0
64000000  | 2000000
64000000  | 4000000

If I remove the code for configuring SWO I only get occasional garbage as output to the telnet client. (Even if I specify all, some or none of the frequencies)

I tried issuing the start command using the debug console, but still didn't see any output. (I could see in the adapter output though that it had received the command correctly)

Marus, 2018-06-28:
Yeah, it's interesting that on that board it doesn't seem to be following the documented J-Link commands for SWO (which do work on several other boards I've used). Do you know what J-Link firmware version and J-Link tools version are being used?

0ge, 2018-06-29:
Really wished I had a different platform to test the JLink on...
I had the same issues before updating the JLink tools to the current version. Here's what I get from the tools involved:
```
~$ JLinkExe
SEGGER J-Link Commander V6.32b (Compiled May  8 2018 18:28:05)
DLL version V6.32b, compiled May  8 2018 18:27:55

Connecting to J-Link via USB...O.K.
Firmware: J-Link OB-SAM3U128 V3 compiled Sep 21 2017 14:14:50
Hardware version: V3.00
[...]
```
```
~$ JLinkExe
SEGGER J-Link Commander V6.32b (Compiled May  8 2018 18:28:05)
DLL version V6.32b, compiled May  8 2018 18:27:55

Connecting to J-Link via USB...O.K.
Firmware: J-Link V10 compiled Apr 20 2018 16:47:09
Hardware version: V10.10
[...]
```
```
~$ JLinkGDBServer
SEGGER J-Link GDB Server V6.32b Command Line Version

JLinkARM.dll V6.32b (DLL compiled May  8 2018 18:27:55)
[...]
```
```
~$ arm-none-eabi-gdb --version
GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 7.12.1.20170417-git
[...]
```

Marus, 2018-06-29:
Well - I might have to see if I can get my hands on one of those boards to see if I can figure the issue out. Will also try those latest J-Link tools with my other boards (haven't been doing much embedded stuff the last couple of months so haven't got to upgrading from the 6.30j release).

0ge, 2018-10-23:
Since VS Code version 1.18.0 (I think) the console is now present, but it receives no information. I've tried various settings in the configuration file, but nothing worked.

closed 2019-06-21 by Marus

ssimek, 2020-07-11:
Hi @Marus,

I'd like to suggest reopening this issue - I have encountered the exact same problem and fixed it by adding the SWO Start command to the startup sequence.

Just guessing, but I'd say that `SWO EnableTarget` tries to configure the TPIU on the core, while `SWO Start` configures SWO on the probe. It worked for me for a long time without `SWO Start`, until I needed to speed things up a bit. After moving from 1 to 3.2 MHz the only way to get console back was through that command.

liligade4, 2022-02-28:
Hi @ssimek ,

Can you help to solve this same problem?
Where you insert the SWO Start command?

Regards,

Liliane

haneefdm, 2022-02-28:
Hi @liligade4, if have not already done so, you may want to read our WiKi document on this.

https://github.com/Marus/cortex-debug/wiki/SWO-Output

Specifically, this following section when your device may be a bit different than normal. 

https://github.com/Marus/cortex-debug/wiki/SWO-Output#swo-configuration

You can customize to your situations but it does help to know what your processor does. SWO is not as simple as it was 5 years ago :-) Not even base addresses can be relied upon.

You can submit a new issue if you think it is not working as expected.

a-cinar, 2022-09-28:
> Hi @ssimek ,
> 
> Can you help me to solve this same problem? Where you insert the SWO Start command?
> 
> Regards,
> 
> Liliane

Just as a reference to other people, I had the same issue on JLink and solved it. 

The `cortex-debug` extension only issues the `SWO EnableTarget` command, which is correct and should be enough according to the [J-Link Debugger's Wiki](https://wiki.segger.com/J-Link_GDB_Server#Supported_remote_.28monitor.29_commands) but somehow it was not enough on its own in order to start the SWO outputs. So I added below to the configurations with my speed settings and it worked:

```json
"postLaunchCommands": [
    "monitor swo start 64000000 875000"
]
```