Marus/cortex-debug #61: OS Awareness issue with OpenOCD

fusedFET, 2018-05-15:
When using an RTOS with a CMSIS-DAP/OpenOCD debug probe, you cannot run the programm as long as the RTOS scheduler is not started. 
Here is my config:
```
{
    "name": "IBDAP",
    "request": "launch",
    "type": "cortex-debug",
    "cwd": "${workspaceRoot}",
    "executable": "./bin/TestProject.elf",
    "servertype": "openocd",
    "configFiles": [
        "${workspaceFolder}/config/debugger/OpenOCD/cmsis-dap.cfg",
        "${workspaceFolder}/config/debugger/OpenOCD/stm32f4x_freertos.cfg"
    ],
    "device": "STM32F407IG",
    "rtos": "FreeRTOS",
    "svdFile": "${workspaceFolder}/config/debugger/STM32F40x.svd"
},
```
The debugger thinks the progamm is either halted, or running once you start debugging. In both cases you cannot resume/pause/restart the programm.

A workaround would be to have a config setting running to run the program to a special symbol different from main or the reset handler, like one of the tasks. This will occur after the scheduler has started.

Marus, 2018-05-16:
Yes, things will not function properly with RTOS support enabled if you try to step through before the scheduler has started, this is a limitation of the GDB servers. 

The process I suggest is to:

1. Set your breakpoints for the areas of concern after the scheduler will have started. 
2. Ensure you do NOT have the runToMain option enabled in the launch configuration
3. Launch the debug session - this should put you at the start of your reset handler. At this point do NOT try to do any debug operations like stepping through code - will result in the GDB server getting in an inconsistent state. 
4. Click the continue button - you should break properly at the first breakpoint, as long as it would be hit after the scheduler was started. From this point you should be able to step normally.

fusedFET, 2018-05-16:
Following that exact process and explicitly setting runToMain as false, does not solve the issue.
The debugger is still not able to continue after step 3. I have attached a log of the gdb session, that might help.
[https://gist.github.com/fusedFET/28a0301d298d7670731c3af47680d7f5](url)

Marus, 2018-05-16:
Can you post your two OpenOCD configuration files - also what version of
openocd are you using.

fusedFET, 2018-05-16:
Sure: The currently used OpenOCD version is 
`GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00487-gaf359c18 (2018-05-12-19:30)`
[stm32f4x_freertos](https://gist.github.com/fusedFET/a45d4d44cda152f9c7b2899c0393e8fe)
[cmsis_dap](https://gist.github.com/fusedFET/aeff1e993cf2fcabd260d123d30ba01b)

Marus, 2018-05-16:
The configure -rtos line was one of the main ones I was wondering about actually (because of how you named the file). 

The Cortex-Debug extension will generate that part automatically when it is configured for RTOS support; I don’t know if running that command multiple times might be causing issues with OpenOCD but you could try without that line and see if that helps. 

If that doesn’t fix it I will see if I can find a way to recreate your issue, as my flow that I mentioned does work with my main setup (although I haven’t tested RTOS stuff much across different setups). What OS are you running VS code on and what version of VS Code are you running?

fusedFET, 2018-05-16:
Using the default file, without the "configure -rtos..." the issue still occurs, but this time the debugger does not find the tasks. I might also try to get a minimal project ready to reproduce the issue.
I am running Win 10: Version 1803, using WSL (ubuntu/bash 4.4.0-17134) as internal vscode console.
VSCode: 1.23.1 and the extension: 0.1.19.

Marus, 2018-05-16:
Yeah, if you can create a minimal version that would be great. You said that it was working in standalone GDB - so I’m assuming you do have the OpenOCD + FreeRTOS helper added to your project (or are using an old enough version of FreeRTOS that it isn’t required) https://github.com/arduino/OpenOCD/blob/master/contrib/rtos-helpers/FreeRTOS-openocd.c

I will try make a setup similar to your environment and see if I can get my minimal example, which does work on my macOS environment, working on that one.

fusedFET, 2018-05-16:
You are absolutely right, I have the last FreeRTOS release `FreeRTOS Kernel V10.0.1` with this helper.

fusedFET, 2018-05-20:
I think I found the reason of my issue. I made an OS abstraction layer, requiring the threads to return a void pointer. For example `static void *MyTask(void *pParams)`. As this does not interfere with FreeRTOS, it will confuse the OS the awareness of OpenOCD. By removing the return value of a void pointer, it does solve the issue. Pretty silly bug on my side.

closed 2018-05-20 by fusedFET

Marus, 2018-05-20:
Thanks for letting me know, as so far I had been unable to recreate the
issue on my side.