Marus/cortex-debug #62: Can't set break point

microwavesafe, 2018-06-07:
I don't seem to be able to set any break points at the moment. It does run to main, but manually adding a break point doesn't seem to do anything and the circle remains an outline. Here is the debug log file of the session, I can't see any obvious errors. [log.txt](https://github.com/Marus/cortex-debug/files/2080428/log.txt)

When I clear the debug console and add / remove a break point I get no output. 

VSCode did update this morning, I'm not sure if this has anything to do with it?

Marus, 2018-06-08:
Will try to look at this later today, might be something that changed in
Visual Studio Code, as I haven’t made any updates in the last couple weeks.

Marus, 2018-06-08:
What version of VSCode are you using? I just tried with both 1.25-insiders and 1.24 release - breakpoints seem to be working for me, 

I did see a similar issue to this with an old insiders build - breakpoints that were set before launch worked fine, but you couldn't change any breakpoints while running (in that case the debug server just didn't receive the event to change the breakpoints from VS Code). Just worried that issue made it into a release - but it doesn't seem to happening to me currently (unless there is an update my machine doesn't want to find yet)

microwavesafe, 2018-06-08:
Here's the version

[image]

So I've done a bit more testing and I can set break points as long as I set them before I launch the debugger. Once the the debugger is launched, all break points appear as circle outlines. I've attached a log of the break points being set before launch, in case it helps.

[log.txt](https://github.com/Marus/cortex-debug/files/2085543/log.txt)

Let me know if I can provide anything else.

microwavesafe, 2018-06-08:
To add to the above, if I set a break point before launch I can then add and remove break points after launch! If I have no break points set before launch, then I can't set any new ones.

Marus, 2018-06-08:
Yeah - that seems like the exact issue I had on an old insiders build - what's odd is that it's not happening to me on the public release build (same version, although mine is the OS X build); also not happening to me on the current insiders build either.

What OS/version are you running on. May see if I can recreate the issue there if it is an os specific thing.

Only other suggestion I can think of is to try and re-install VS Code and/or the extension - in case something got broken in an upgrade process somewhere.

xiaoyongdong, 2018-06-09:
I have the same problem. My OS version is Windows10 x64 and VS Code is 1.24.0.

Marus, 2018-06-09:
Ok - will try to look into this a bit more to see if I can re-create the issue. While I don't expect it is related (as I'm guessing it's just not getting the commands from VS Code at all), can you send me your launch.json configuration(s)

Marus, 2018-06-11:
Hi @microwavesafe and @xiaoyongdong 

So far I've still had no luck recreating this. I've tried on my Windows 10 and Ubuntu (17.10 - haven't had a chance to setup a 16.04 VM yet) with VSCode 1.24.0. In both cases don't seem to have any issues setting breakpoints in my sample programs while running (or paused).

In addition to your launch.json configurations can you send me a list of other extensions that you have and any relevant user or workspace settings. Perhaps there is some combination that is causing issues.

xiaoyongdong, 2018-06-11:
<br>Hi @Marus</br>
<br>Here is my all settings:</br>
[launch.txt](https://github.com/Marus/cortex-debug/files/2088505/launch.txt)
[settings.txt](https://github.com/Marus/cortex-debug/files/2088507/settings.txt)
[user_settings.txt](https://github.com/Marus/cortex-debug/files/2088510/user_settings.txt)</br>

<br>The following is all extensions: </br>
<br>Auto Close Tag</br>
<br>Auto Rename Tag</br>
<br>Bracket Pair Colorizer</br>
<br>C/C++</br>
<br>C/C++ Clang Command Adapter</br>
<br>C/C++ Snippets</br>
<br>Cortex-Debug</br>
<br>hexdump for VSCode</br>
<br>Intel HEX format</br>
<br>One Dark Pro</br>
<br>Path Intellisense</br>
<br>Python</br>
<br>Verilog HDL</br>

microwavesafe, 2018-06-11:
Last idea, is there a way I can drop a modified file manually into your extension on my machine, that provides more debug output that I can send you. I'm happy to give this a go if you think it will help.

Marus, 2018-06-11:
Shouldn't be any difference between uninstall and disable as long as you do the reload operation after disabling.

I'll see if I can make a custom build that may get some additional information.

xiaoyongdong, 2018-06-12:
I remember the last version before resolved #56 work well.

skalldri, 2018-06-13:
I think I have the same issue. I'm using v0.1.20 of the Cortex-Debug extension. When I launch the debugger, it connects to my MCU: I can "pause" execution and it works correctly. If I try to set a breakpoint it just gives me the "unverified breakpoint" grey circle instead of the red circle to indicate that the breakpoint is set.

However, if I set a breakpoint before I launch the debugger, then the breakpoint is set correctly (red circle) and I can then set more breakpoints after launching the app.

If I remove all the breakpoints and re-launch the debugger, it repro's the original issue (unverified breakpoints for all breakpoints).

Interestingly, the debugger is still breaking into the reset handler on first boot. If I set a breakpoint manually through the GDB console in VSCode, those breakpoints will hit normally, regardless of whether or not the GUI breakpoints are working.

I've turned on extension debugging through "showDevDebugOutput":true in my launch configuration. When the bug repros, I see no debug output at all when I try to set a breakpoint through the GUI. When The bug is not repro-ing (IE: I launch the debugger with a breakpoint already set) then I get some debug spew that matches the location that I set the breakpoint.

Marus, 2018-06-13:
Hi,

I just posted a v0.1.21-test build on GitHub as a VISX file. This has a bit more logging that will hopefully help me determine what the issue is. You can run it with or without showDevDebugOutput enabled - as this logging isn't disabled without that flag.

xiaoyongdong, 2018-06-14:
@Marus I installed  the test version of the extension you post and here is the log file
[log.txt](https://github.com/Marus/cortex-debug/files/2100585/log.txt)

xiaoyongdong, 2018-06-14:
I think when bind the event 'debug-ready', it already emit.

Marus, 2018-06-14:
Thanks @xiaoyongdong 

Will test out that PR and release tonight. Hopefully that fixes for everyone.

Marus, 2018-06-15:
I've made a pre-release version with the proposed fix from @xiaoyongdong - since I wasn't able to recreate the issues on my computers it would be good if you could test it and see if it resolves the problems for you before I do a general release with it.

https://github.com/Marus/cortex-debug/releases/tag/v0.1.21-test2

microwavesafe, 2018-06-15:
I have installed the pre-release and can confirm it works with break points set after launching and still works with break points set during run and before launch. It looks like it works to me.

Marus, 2018-06-15:
I've submitted v0.1.21 to the marketplace. Thanks @microwavesafe, @skalldri for your testing and patience while we tried to sort this out, and thanks @xiaoyongdong for your assistance in creating the fix.

closed 2018-06-15 by Marus