Marus/cortex-debug #69: support for more vscode variables

lemrey, 2018-06-19:
It'd be great if we could use more variables to specify the path of the executable. Visual Studio Code has many [variables](https://code.visualstudio.com/docs/editor/variables-reference) but only the (deprecated) ${workspaceRoot} seems to be expanded.

AJAnderson, 2018-06-23:
Second this, I could get a lot of utility out of ${workspaceFolderBasename}

Marus, 2018-06-28:
Will look at adding some of the additional variables that make sense in the next release, that will likely be around July 6-7.

PhilippHaefele, 2021-11-16:
I think in the meanwhile all variables should work (maybe VSCode changed something).
@haneefdm I think we can close this one here

PhilippHaefele, 2021-11-16:
The only thing to mention here, is that variable substitution doesn't work for settings.json in manner it does for launch.json (https://code.visualstudio.com/docs/editor/variables-reference#_is-variable-substitution-supported-in-user-and-workspace-settings). Also i'm not sure if `${workspaceFolderBasename}` does work there right now (haven't found something in code). Most settings are available in launch.json, so user should use them when variables are needed.

haneefdm, 2021-11-16:
The variables you can use in launch.json are documented here (like @lemrey mentioned)

https://code.visualstudio.com/docs/editor/variables-reference#_predefined-variables

Cortex-Debug has no plans to introduce new variables. Note that you can create your own in settings.json (don't have to be any extension related) and use them in launch.json like this

```
 ${config:abc.xyz}
```
where `abc.xyz` is some variable defined in the User settings or Workspace/Folder settings

And no, settings.json is also something we will not mess with. This is controlled/modified by VSCode and the user and we don't intend to create a second/parallel settings infrastructure/API