Marus/cortex-debug #70: Cortex Peripheral window not showing what is actually in ram

stevenj, 2018-06-19:
I am trying to debug a Timer Counter, so was using the Cortex Peripheral view on my SAMD21 board.  And it dodn't look right, so i dumped memory at that address, and yes, a raw memory dump is showing the expected information.  Its like the Cortex Peripheral view has stopped reading from memory?

Screenshot of the problem.  [http://i.imgur.com/Fqj2ybs.png]()

You will see the Timer peripheral base address and memory base address match, but the timer peripheral display is not the same data.

diegoherranz, 2018-07-09:
Same here with an STM32F103C8.
Trying to check some GPIO registers and they are never changing. However, reading the registers directly shows changing values.

Starting to use this plug-in today. Looks really nice! Thanks for the work here.

mike-rob, 2018-07-12:
I have seen the same thing as well. Not sure how I did it but browsing the Cortex peripherals tree (maybe collapsing it?) made it refresh and then it was correct. Maybe this is a refresh issue?

windwolfreal, 2018-08-19:
The same problem with STM32F103VB.

filiporlo, 2018-10-01:
The same STM32F103C8
Changing using "CORTEX PERIPHERALS" is not sync in this windows(but change is done), when I add watch for example GPIOA new value is show in watch.

mike-rob, 2018-10-03:
Double clicking on a top level item (Register Group) will trigger a read of the group.  This can be a good thing because reading from registers can have side effects.

jaxxzer, 2018-10-13:
+1 on stm32f103rc

jaxxzer, 2018-10-13:
+1 on stm32f103rc

Travis-Spidertracks, 2018-10-24:
+1. On NXP LPC54616 if I double click a top level register and then execute an instruction in updates, otherwise it is out of sync

thepartisan, 2018-12-12:
+1 same behaviour on STM32F072RB with both openocd  (0.10.0+dev-00587-g63fcef493) and ST-Util 1.3.0

chriswils, 2018-12-19:
Same with nRF52840 with provided .svd

pmaxim25, 2018-12-20:
+1 on STM32F103 and STM32L053

Marus, 2019-01-19:
Will start looking into this - any time execution stops it should trigger the view to re-read the memory of any registers that have been expanded - but perhaps something is causing the steps to not trigger that re-read.

Another thing to consider - if it is peripheral that continues to run even while the main core has stopped then the view will only be up-to-date with the information that was in the registers when execution stopped - not reflecting any changes after the stop occurred.

For STM32 (don't know the nRF or LPC chips well enough to say with any certainty) depending on how you have configured the DBGMCU peripheral will have an impact on if Timers and other peripherals that depend on the timers continue to run while the core is stopped.

Marus, 2019-06-21:
We think that updating of the data should be working much more reliably now, if issues persist please open a new issue.

closed 2019-06-21 by Marus