Marus/cortex-debug #72: Setting Breakpoints in VSCode

mcprakash, 2018-06-27:
Hi 
 I really love this application and it works great on Windows. I really need your help to help me with setting Breakpoints and watching variables.

 I am using a STM32 Nucleo board (works with OpenOCD) and somehow the stlink does not work well. Regardless, I am unable to watch any variables and set breakpoints (on linux). On windows, I am able to set the breakpoints but the hardware just does not stop at the breakpoint. 

Best
Chandra

chetferry, 2018-06-28:
As you are on a Nucleo board, you can re flash the debugger to be a JLink.  Here is the link with instructions:

https://www.segger.com/products/debug-probes/j-link/models/other-j-links/st-link-on-board/

You will need to use  Windows OS to run the utility that changes the on board STLink to be a JLink. But then it should be good on your Linux OS.  I am using the cortex-debug extension on MacOS.  No problems with Setting breakpoints and watching variables.  I am also using a Nucleo board that I have re flashed to be a JLink.

You will also need to install the JLink tools on you Linux computer.  The files to do this can be found here:

https://www.segger.com/downloads/jlink/#J-LinkSoftwareAndDocumentationPack

Here are the contents of my launch.json file:

{
    "version": "0.2.0",
    "configurations": [
    
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/f767_rtos_test.elf",
            "svdFile": "${workspaceRoot}/STM32F7x7_v1r2.svd",
            "name": "Debug (J-Link)",
            "device": "STM32F767ZI",
            "interface": "swd",
            "ipAddress": null,
            "serialNumber": null
            
        }
    ]
}

Cheers
Chet

mcprakash, 2018-06-28:
Hi @chetferry 
 Thanks for the info to change it into a J-Link. My question is, if I change it from an ST-LINK to J-LINK, how do I program the chip. I am currently using ARM MbedOS to compile my application. So, how would I be able to flash the .bin file. 

Thanks
Chandra

Marus, 2018-06-28:
While ST-Link should in theory work (either with St-Utils or OpenOCD), I've found flashing it to be J-Link to be much more reliable and have better performance, so in general for most people I recommend that (if you are just using it on the Nucleo/Discovery boards, by the license you are not supposed to use it for off-dev-board development). 

As for programming your target after flashing, if you are using Cortex-Debug with a "launch" type (versus attach), then Cortex-Debug will take care of sending the appropriate commands to GDB to flash the ELF format executable that the launch.json file should be pointing to. I unfortunately haven't worked with the Mbed tools any significant amount, but to my knowledge they have export to makefile type projects, I expect that would allow you to compile an ELF format executable which could be used with Cortex-Debug and that would be able to flash directly through the launch.

chetferry, 2018-06-28:
I also agree that JLink is much more solid and reliable in general.  I have used the ST Link quite a bit in the past with a few toolchains, and have always had a better experience with JLink.  If you are doing development for your day job it is well spent money to purchase a JLink debugger so you have access to the Segger tools.  If all your development will be on the Nucleo board then you can use the JLink tools for free.

Marus, 2018-06-29:
Yeah, so if you are using the launch type in your launch configuration pointing to their generated ELF file the Cortex-Debug tools will automatically do the following on launch:

* Halt the target
* Program the target with the firmware in the ELF file
* Reset the target
* And if runToMain is enabled it will then set a temporary breakpoint at "main" and start the processor waiting for it to run to main.

As @chetferry said, I think the J-Link tools are just significantly better than ST-Link. If you are doing commercial work I think they are well worth the cost. For just on the nucleo board you can do the flash; if you are doing non-commercial work for your own hobby/learning then you also have the option of the J-Link EDU Mini (~$20) and the J-Link EDU (~$70), both of which are quite good, especially for the price. 

As to why the breakpoints wouldn't be working on your Linux machine, I'm not sure. I have used the tools successfully (at least for basic testing) on Ubuntu (17.10 and 18.04) with OpenOCD and ST-Utils connecting to ST-Link and J-Link. There were some issues a while back with some issues setting breakpoints after the latest VS Code update - these seem to be fixed in the last update of Cortex-Debug.

chetferry, 2018-06-29:
Marus,

I am using FreeRTOS, JLink, MacOs (High Sierra), CMake Tools 1.0.1 vscode plugin, and gcc-arm-none-eabi-7-2017-q4-major for my compiler.  I tried to set the runToMain property to true in my launch.json file.  Here is my launch.json file:

{
    "version": "0.2.0",
    "configurations": [
    
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/f767_rtos_test.elf",
            "svdFile": "${workspaceRoot}/STM32F7x7_v1r2.svd",
            "name": "Debug (J-Link)",
            "device": "STM32F767ZI",
            "interface": "swd",
            "ipAddress": null,
            "serialNumber": null,
            "rtos": "FreeRTOS",
            "runToMain": true
        }
    ]
}

When I launch a debug session this is what I get at the debug console:

Reading symbols from /Users/chet/Dropbox/vscode_c_code/f767_rtos_test/build/f767_rtos_test.elf...
done.
0x08001758 in prvCheckTasksWaitingTermination () at /Users/chet/Dropbox/vscode_c_code/f767_rtos_test/Middlewares/Third_Party/FreeRTOS/Source/tasks.c:3376
3376			while( uxDeletedTasksWaitingCleanUp > ( UBaseType_t ) 0U )
Not implemented stop reason (assuming exception): undefined
Resetting target
Resetting target
[New Remote target]

Everything is frozen and I cannot proceed with debug from here.

If I change my launch.json back to as shown below:

{
    "version": "0.2.0",
    "configurations": [
    
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/f767_rtos_test.elf",
            "svdFile": "${workspaceRoot}/STM32F7x7_v1r2.svd",
            "name": "Debug (J-Link)",
            "device": "STM32F767ZI",
            "interface": "swd",
            "ipAddress": null,
            "serialNumber": null
        }
    ]
}

The debug starts successfully in my startup_stm32f767xx.s  as shown here:

[image]

I can then hit run and can get to my first breakpoint in main, I can debug in my main code until the scheduler starts.  I can run to my first breakpoint after the scheduler is running inside a task and all is well and seems to work fine.  I do get this output at the debug console when I start as described and I am halted at my first breakpoint inside an RTOS task:

Reading symbols from /Users/chet/Dropbox/vscode_c_code/f767_rtos_test/build/f767_rtos_test.elf...
done.
0x08001758 in prvCheckTasksWaitingTermination () at /Users/chet/Dropbox/vscode_c_code/f767_rtos_test/Middlewares/Third_Party/FreeRTOS/Source/tasks.c:3376
3376			while( uxDeletedTasksWaitingCleanUp > ( UBaseType_t ) 0U )
Not implemented stop reason (assuming exception): undefined
Resetting target
Resetting target
Breakpoint 
2, main () at /Users/chet/Dropbox/vscode_c_code/f767_rtos_test/Src/main.c:93
93	  HAL_Init();
Breakpoint 
1, StartDefaultTask (argument=0x0) at /Users/chet/Dropbox/vscode_c_code/f767_rtos_test/Src/freertos.c:120
120	        SEGGER_RTT_printf(0, "What about Bob! : %d \r\n", BobCounter);

So it complains but seems to work just fine for my purposes.  I do need to use RTOS, and I often have some initial code that must run before the scheduler starts.  Just thought I would provide some feedback in case it helps.

Thanks.

closed 2021-03-22 by mcprakash