Marus/cortex-debug #73: Feature Request: Allow Additional Server Args for OpenOCD

montaguk, 2018-06-29:
Currently, openocd.ts allows limited control over the arguments that get passed to the openocd server at launch.  As best I can figure, this is done via the configFiles and searchDir options in the launch configuration.

I have a unique case where I would like to launch OpenOCD with some custom commands, but it does not appear that this is currently possible.  I plan to implement support for this soon on a fork as I can't get my boards to talk without the extra commands.

I had two thoughts for how to go about this, and wanted to get some feedback before diving in, in the hopes that I could merge my changes back into the mainline via a pull request.

First option would be to add another launch configuration argument called cmdList or similar, that would take an array of command strings and output a single string of formatted commands to pass to OpenOCD.

The second option is to add a new launch configuration argument called args or similar, that simply allows the user to directly enter a string of extra arguments that would pass through to the OpenOCD server directly.  This would allow more flexibility in the long term, but also could be a little less user friendly.

I'm also open to any other approach/ideas, as I'm really just looking for the shortest path to getting a functional development environment for my boards.  If I've overlooked something that can get me there without a code change, then I'm all for it.

Thanks!
-Kyson

Marus, 2018-06-29:
I certainly have no issues with adding some support for additional command line arguments or commands to be passed to OpenOCD (or other servers). Personally I think doing it at the command level (so people don't have to get as much formatting correct) is probably better than raw command line arguments.

I would probably prefix the launch configuration property with something to be clear that it is OpenOCD only. So maybe something like openOCDLaunchCommands.

If you need additional commands to be sent at the GDB level there are the preLaunchCommands/postLaunchCommands (and attach versions) that are available.

I can probably implement this for my next release (this will likely be around July 7), or if you want to try implementing yourself before then I would certainly be happy to take a pull request.

BarrettLowe, 2020-02-13:
It may be nice to do this for the jlinkgdb server as well. I want to manually set the adapter speed (for example).

haneefdm, 2020-02-13:
The new feature to allow server args is already merged into the codebase. Expect it in the next release.  (already available in the pre-release). It applies to all gdb servers.

closed 2020-02-13 by haneefdm