Marus/cortex-debug #75: Attach to running code

mike-rob, 2018-07-12:
Using the J-link and eclipse I can attach a debugging session to code already running on a chip. Would it be possible to add this option to you extension? I think basically it connects but skips the programing step. It normally has a name like `Attach only` or `Attach to a running target`

I really like your extension as it allows me be able to move away from heavy and buggy eclipse biased IDEs. :+1:

Marus, 2018-07-13:
You can already do this - in your launch.json launch configuration use `"request": "attach"` instead of `"request": "launch"`.

A few other options may need to change as well - as they may not be applicable to attach vs. launch type requests (for example "runToMain" doesn't exist in attach), and the pre/post LaunchCommands are renamed to pre/postAttachCommands.

mike-rob, 2018-07-19:
Thanks, did not see that. Thanks! :+1:

closed 2018-07-19 by mike-rob