Marus/cortex-debug #77: SWV not working

n-is, 2018-07-20:
I can't get SWV to work with this extension. I get the output in ST-LINK Utility program's SWO-Viewer, but not in OUTPUT console of VS Code. Is there something wrong with my launch.json file or should I look for error in code. When I ran the program in single-step mode and just send a char using ITM_SendChar() function, the character is sent, but I can't see it in the console. Sorry for posting my problem here.

This is my launch.json file.

`{
	"configurations": [
		{
			"cwd": "${workspaceRoot}",
			"executable": "./_target/hello_world.elf",
			"name": "Debug Microcontroller",
			"request": "launch",
			"type": "cortex-debug",
			"servertype": "openocd",
			"configFiles": [
				"C:\\OpenOCD-20170821\\share\\openocd\\scripts\\board\\stm32f4discovery.cfg"
                        ],
                        "svdFile": "${workspaceRoot}/_target/system/STM32F407.svd",
                        "preLaunchCommands": [
                                "monitor reset halt",
                                "tbreak main"
                        ],
                        "interface": "swd",
                        "swoConfig": {
                                "source": "probe",
                                "enabled": true,
                                "swoFrequency": 2000000,
                                "cpuFrequency": 168000000,
                                "decoders": [
                                        {
                                                "port": 0,
                                                "label": "Output"                                                                                               
                                        }
                                ]
                                
                        }
                        
		}
	]
}`

diggit, 2018-08-27:
Hi @n-is, where can I find this drop-down box?
I can't see it anywhere.

Do you see any problem in my config?

```json
{
    "preLaunchTask": "build",
    "type": "cortex-debug",
    "request": "launch",
    "name": "Debug (OpenOCD)",
    "showDevDebugOutput":true,
    "servertype": "openocd",
    "interface": "swd",
    "cwd": "${workspaceRoot}",
    "executable": "${workspaceRoot}/main.elf",
    "device": "STM32F746IG",
    "svdFile": "${workspaceRoot}/STM32F7x6.svd",
    "configFiles": [
        "${workspaceRoot}/openocd.cfg",
    ],
    "swoConfig":{
        "enabled":true,
        "source":"probe",
        "swoFrequency":2000000,
        "cpuFrequency":200000000,
        "decoders": [
            {
                "port": 0,
                "label": "SWO output",
                "encoding":"ascii"
            }
        ]
    }
},
```

diggit, 2018-09-01:
You are right. Now ti works. Thanks!

closed 2019-06-21 by Marus

ahmadshokati, 2025-04-17:
Did you use St-Link debugger for this purpose?

haneefdm, 2025-04-17:
NB: SWO/RTT are now in the TERMINAL . They moved from OUTPUT to TERMINAL 3 years ago.

@ahmadshokati They used Cortex-Debug with openocd for it. You have to download the latest SW for full support using ST-Link tools (gdb-server). Openocd, ST-Link and ST-Util should all work. I recommend ST-Link with our latest release. OpenOCD comes close second.

https://github.com/Marus/cortex-debug/releases

ahmadshokati, 2025-04-17:
@haneefdm I've downloaded the latest release, but I'm still unable to print SWO data. I've tried both ST-Link and OpenOCD in my launch.json, but no luck so far.

diggit, 2025-04-17:
Hi @ahmadshokati, please don't [necrobump](https://www.urbandictionary.com/define.php?term=necrobump) this issue and create new one with all details like your debug configuration, logs, version info, etc.