Marus/cortex-debug #82: Can increase the pyocd custom parameters?

lgl88911, 2018-08-24:
I want to set pyocd param "-f", "--frequency", help = Set the SWD clock frequency in Hz." for debug. Is there a plan? Thanks

haneefdm, 2021-07-10:
Sorry for the late response.

Support for adding your own arguments is already supported via `serverArgs` in launch.json.

However SWO is not directly supported for pyocd. I will change the topic to reflect needing support for SWO, if you don't mind.

lgl88911, 2021-07-26:
It's really late, so much so that I have given up using SWD. Fortunately, it is now supported. I will try it when I am free, thank you.
The new title looks good, and this modification is indeed for pyocd.

haneefdm, 2021-07-26:
I think you meant SWO. Thanks, try it out when you have a chance. I am closing this for now and pyocd based SWO is now released

closed 2021-07-26 by haneefdm