Marus/cortex-debug #83: Feature Request: customization of executable name for gdb

d0m72, 2018-08-24:
Apparently cortex-debug tries to locate an executable named **arm-none-eabi-gdb.exe** when launching a debug session. Unfortunately in my case my Arm toolchain is Linaro, and this last has **arm-eabi-gdb.exe** in \bin folder. So I did something a bit ugly, by copying and renaming this exe to be compliant with cortex-debug calling requirements. 

I think that, similar to the setting **cortex-debug.armToolchainPath** we could have a setting to override this executable name which would be set by default with actual name. 

Apart from that, it works great with texane stlink v1.3 gdb server and an STM32476VG eval board. 

A big thank for your efforts !

higaski, 2018-09-26:
I second that. That would be a great feature. I pretty much have the same problem except I need to use the gdb from the xtensa-esp32 toolchain.

nickray, 2018-12-07:
On Debian, GDB is called `/usr/bin/gdb-multiarch` now.

nickray, 2018-12-07:
Isn't https://github.com/Marus/cortex-debug/blob/master/src/common.ts#L128 exactly this feature? Why hide it?

Marus, 2019-01-19:
Will look into options for customizing the executable names

Marus, 2019-01-22:
There is a hidden launch.json configuration variable `gdbpath` that takes a full path to a the GDB executable - this may allow you to work around this at the moment - although I'm not sure it will work in all cases. You would still require `arm-none-eabi-objdump` though for full functionality as that is used to read the symbol table (this should be on the system path or the toolchainPath setting).

farrenv, 2019-05-20:
@Marus     Would this provide a path to using cortex-debug to have GDB connect to a GDB server already running on a remote target?

aunsbjerg, 2019-06-02:
I'm in need of this feature as well. I'm trying to use vscode + cortex-debug as an IDE for zephyr, having moderate success so far. Zephyrs SDK has a custom version of the arm toolchain where all binaries have the following prefix `arm-zephyr-eabi-`. I'm thinking that instead of or in addition to having custom gdb binary names, the toolchain prefix could be customized so that all binaries (gcc, gdb, objdump, etc) would be prepended with this prefix.

My workaround right now is using `armToolChainPath` to point to the zephyr SDK arm toolchain combined with creating a `arm-none-eabi-gdb` symlink that points to `arm-zephyr-eabi-gdb`. Works fine, but is a bit of a hassle, especially if setting up on multiple dev computers.

MBetters, 2020-06-04:
Just confirmed that setting `"cortex-debug.gdbpath": "/usr/bin/gdb-multiarch"` in `.vscode/settings.json` does not work. Any progress on this issue? Awesome extension btw!

diegoherranz, 2021-01-23:
The hidden configuration variable `gdbpath` was working for me up to version 0.3.7.
On version 0.3.8 or 0.3.9 it complains that `/usr/bin/arm-none-eabi-gdb` cannot be found.

It would be very useful to have official support for a configuration variable `gdbpath` or even try to locate `gdb-multiarch` if `arm-none-eabi-gdb` is not available, given that seems to be the default on Debian-based systems.

Thanks!

ryankurte, 2021-01-25:
oops, updated the plugin and `gdbPath` no longer works for me either, second that it'd be great to have this back!

Marus, 2021-01-25:
Do apologize about breaking the work around. I'm actually working on adding a proper configuration setting for this now - should be out in the next release - which will hopefully be out by sometime tomorrow.

Marus, 2021-01-25:
Sorry - this apparently got left out of the release notes for 0.3.8 - but a user setting (settings.json) for gdbPath was introduced - allowing you set this at the user level. 

Inadvertently this broke the undocumented `gdbpath` in launch.json - this was unintentional.

The configuration property should be `"cortex-debug.gdbPath"`

For 0.3.10 I'll be re-adding support for specifying this in the launch.json as well - although likely under the `gdbPath` key for consistency with the user setting (I believe the previous undocumented workaround was with `gdbpath` as the key).

closed 2021-01-25 by Marus

diegoherranz, 2021-01-26:
Thanks for improving this!