Marus/cortex-debug #85: Serial problem

diggit, 2018-08-26:
Hi, I am trying to use SWO on BMP.

Running vscode 1.26.1-2 (archlinux AUR)

Whenever I try to start debug with swo, this error pops up:
`Unable to load Serial Port Module. A recent Visual Studio Code update has likely broken compatibility with the serial module. Please visit https://github.com/Marus/cortex-debug for more information.`

I noticed, that binary modules submodule is quite outdated.

Or is there something wrong in my config?

```json
{
	"preLaunchTask": "build",
	"type": "cortex-debug",
	"request": "launch",
	"name": "Debug (BlackMagic)",
	"showDevDebugOutput":false,
	"servertype": "bmp",
	"cwd": "${workspaceRoot}",
	"executable": "${workspaceRoot}/main.elf",
	"device": "STM32F746IG",
	"svdFile": "${workspaceRoot}/STM32F7x6.svd",
	"BMPGDBSerialPort": "/dev/ttyACM0",
	"targetId": 1,
	"swoConfig":{
		"enabled":true,
		"source":"/dev/ttyACM1",
		"swoFrequency":2000000,
		"cpuFrequency":200000000,
		"ports":[
			{
				"number":0,
				"label":"text",
				"format":"console"
			}
		]
	}
}
```

Marus, 2019-01-20:
Yes - the serial stuff is unfortunately out of date from VS Code runtime changes - trying to get those working again now - but it hasn't been easy.

Marus, 2019-01-21:
This should now be fixed on macOS and Linux if you are running the new 0.2.2 release (requires VS Code 1.29 or higher); Windows support still requires some additional work to get the binary modules compiling for this version.

diegoherranz, 2019-02-16:
I'm getting this same error on vscode 1.31, on Linux.
If I downgrade to vscode 1.29.1, it works OK. Maybe it needs updating binary modules again?

Thanks!

diegoherranz, 2019-12-09:
Still getting the same error on the following config:

> Name: Cortex-Debug
> Id: marus25.cortex-debug
> Description: ARM Cortex-M GDB Debugger support for VSCode
> Version: 0.3.4
> Publisher: marus25
> VS Marketplace Link: https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug

VS code info

> Version: 1.40.2
> Commit: f359dd69833dd8800b54d458f6d37ab7c78df520
> Date: 2019-11-25T14:54:40.719Z
> Electron: 6.1.5
> Chrome: 76.0.3809.146
> Node.js: 12.4.0
> V8: 7.6.303.31-electron.0
> OS: Linux x64 4.4.0-170-generic

Any chance this can be looked at? I find the extension very useful and it would be nice to be able to use this.

Thanks!

diegoherranz, 2019-12-10:
Even if it's just pointing at the right direction, maybe I can help with this.
Thanks!

haneefdm, 2019-12-10:
I wish I could help. I don't have a BMP setup. I will try to fake one to see what modules are out of date, but I will first try on a Mac.

I am not sure how to go about setting up a debugging setup similar to yours. I have an STLink V2 [Nucleo board](https://www.st.com/en/evaluation-tools/nucleo-f429zi.html). Would that be on any use?

haneefdm, 2019-12-10:
@diegoherranz  Have you tried updating the binary modules yourself in your install copy of the extension?  Since I don't have the right HW, I can only try so much.

haneefdm, 2019-12-10:
@diegoherranz Could you try the following to try to install serialport yourselves. Let me know if it works.

For Line # 1, Use the proper version of cortex-debug
For Line # 2, the directory name is vscode-node-version/platform/arch

- vscode-node-version can be found by doing an 'About VSCode' in the VSCode menu
- platform is either 'linux' or 'darwin' or 'win32'
- arch is either 'x64' or 'ia32'

For Line # 3, I am not sure if you need to do a 'sudo'. I generally don't
For Lines # 4 and # 5, you can do it other ways. Idea is to install serialport it in a fresh directory and then move it or it will use the package.json a few directories above and install serialport in there
```bash

1. cd ~/.vscode/extensions/marus25.cortex-debug-0.3.4/binary_modules
2. mkdir -p v12.4.0/linux/x64
3. cd v12.4.0/linux/x64
4. (mkdir -p /tmp/Test ; cd /tmp/Test ; npm install serialport)
5. mv /tmp/Test/node_modules .
```
Restart VSCode

diegoherranz, 2019-12-10:
Many thanks for your help.

I don't manage to make it work.

I've followed all your steps. And here you can see the structure after I copied the node_modules folder obtained in step 4:

```
~/.vscode/extensions/cortex-debug/binary_modules$ tree  -L 5
.
├── v12.4.0
│   └── linux
│       └── x64
│           └── node_modules
│               ├── ansi-regex
│               ├── aproba
│               ├── are-we-there-yet
│               ├── bindings
│               ├── bl
│               ├── chownr
│               ├── code-point-at
│               ├── console-control-strings
│               ├── core-util-is
│               ├── debug
│               ├── decompress-response
│               ├── deep-extend
│               ├── delegates
│               ├── detect-libc
│               ├── end-of-stream
│               ├── expand-template
│               ├── file-uri-to-path
│               ├── fs-constants
│               ├── gauge
│               ├── github-from-package
│               ├── has-unicode
│               ├── inherits
│               ├── ini
│               ├── isarray
│               ├── is-fullwidth-code-point
│               ├── mimic-response
│               ├── minimist
│               ├── mkdirp
│               ├── ms
│               ├── nan
│               ├── napi-build-utils
│               ├── node-abi
│               ├── noop-logger
│               ├── npmlog
│               ├── number-is-nan
│               ├── object-assign
│               ├── once
│               ├── prebuild-install
│               ├── process-nextick-args
│               ├── pump
│               ├── rc
│               ├── readable-stream
│               ├── safe-buffer
│               ├── semver
│               ├── serialport
│               ├── @serialport
│               ├── set-blocking
│               ├── signal-exit
│               ├── simple-concat
│               ├── simple-get
│               ├── string_decoder
│               ├── string-width
│               ├── strip-ansi
│               ├── strip-json-comments
│               ├── tar-fs
│               ├── tar-stream
│               ├── tunnel-agent
[... 249 lines cut ...]

I'm using the following config in `launch.json`:

```
"swoConfig": {
                "enabled": true,
                "cpuFrequency": 72000000,
                "swoFrequency": 4000000,
                "source": "/dev/ttyUSB1",
                "decoders": [
                    {
                        "type": "console",
                        "port": 0,
                        "showOnStartup": true
                    }   
                ]
            }
```

I'm using this FTDI-based adapter (https://github.com/diegoherranz/steppenprobe), but
I believe you can test this error with any adapter since this is using an external serial port (/dev/ttyUSB1 in my case) instead of one built with the probe. Even if you didn't have any serial port, it should be able to load the `serialport` module and then fail to find that specific serial port, I guess.

I've spent a couple of hours testing different nodejs versions and similar stuff, but my knowledge of all this is limited so I've had no success.

Thanks!

Marus, 2019-12-11:
I'm actually working on updating all the binary modules for the serial port
based ITM support at the moment. Still need to get the Windows versions of
the binary modules built. That should hopefully get it working until the
next time VSCode updates the runtime. Hopefully can at least get a test
version out this weekend.

haneefdm, 2019-12-12:
@Marus Could we automate this? I mean have some scripts for each platform.

We could even do it on a user's computer rather than bundling with the extension? We could check each time the extension is launched and attempt to install the module if it is missing/out-of-date. This way, VSCode could update and we don't have to release an extension update. Food for thought. And if the install fails, still no harm done as the extension is still usable for those that don't need a serial port.

haneefdm, 2019-12-12:
FYI: Others running into this same issue and their solution

https://cultivatehq.com/posts/how-we-built-a-visual-studio-code-extension-for-iot-prototyping/

@diggit the problem is that the `serialport` module has to be bound to the particular version of `electron` being used by VSCode. That is why what I said does not work.

Marus, 2019-12-12:
@haneefdm - unfortunately there is not a way currently to have the install of the modules done at install time on the users computer (I have been slowly working on setting up automating building the modules on some VMs here - so at least what I have to do is simpler). 

I know there has been some talk of VSCode allowing post-install steps with apm/npm to do this type of thing - but currently nothing has come of that. Still hopeful it will eventually come-to-pass.

Hope to have an update for the latest versions out this weekend - to at least get running for the time being.

diegoherranz, 2019-12-12:
Thank you both for taking care of this.

diegoherranz, 2020-01-22:
Did you get a chance to work on this, @Marus?
Thanks!!

haneefdm, 2020-02-09:
I made a PR #245 with the new serialport added. I think we can close this once it is released.

closed 2021-02-14 by Marus