Marus/cortex-debug #86: How to save SWO data to a file?

n-is, 2018-08-31:
I would like to redirect the SWO output to a file. Is there any way I can do this?

perlindgren, 2019-01-04:
You can use postLaunchCommands to setup `openocd` to output to a file (or external ITM viewer). This bypasses the cortex-debug plugin, so your launch config should not have any SWO related fields.

Best regards,
  Per

n-is, 2019-01-05:
@perlindgren Thank you for the reply. Could you possibly provide some examples? Thanks again.

perlindgren, 2019-01-05:
Hi.
The particular setup is for Rust (but I assume it would work the same in C/C++), nothing language dependent here. The file `/tmp/itm.log` will contain the raw ITM packages. If you use `tail -f /tmp/itm.log` you will get readable output - but its just luck, as the terminal/console actually prints the header (but the characters are "unprintable" so no visual output (at least under arch-linux). 

A better option is to use the Rust tool `itmdump` (just `itm` on crates.io). It filters out the header and passes on the decoded stream. In that case you may use the `/tmp/itm.log` as a fifo (zero sized file). `mkfifo /tmp/itm.log`, before starting `itmdump -f /tmp/itm.log -F`. See the `itmdump` docs for further info. (FIFOs might not work on all platforms, but should be ok under linux/OSX.)

Here is an example launch config.
```
 {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "openocd",
            "name": "itm 16Mhz /tmp/itm.log (debug)",
            "preLaunchTask": "cargo build --example itm",
            "executable": "./target/thumbv7em-none-eabihf/debug/examples/itm",
            // uses local config files
            "configFiles": [
                "./stlink.cfg",
                "./stm32f4x.cfg"
            ],
            "postLaunchCommands": [
                // sets the core clock to 16000000 (see ./stm32f4x.cfg)
                "monitor reset init",
                // 16000000 must match the core clock frequency
                "monitor tpiu config internal /tmp/itm.log uart off 16000000",
                // enable ITM port 0
                "monitor itm port 0 on"
            ],
            "cwd": "${workspaceRoot}"
}
```
Let me know if this works for you.
/ Per

perlindgren, 2019-01-20:
> This is a good idea - can look into redirecting SWO output to a file (either channel specific or as a whole) - not currently in place. Depending on your particular debug probe you may be able to use the example postLaunchCommands that @perlindgren<https://github.com/perlindgren> posted - although that would likely prevent cortex debug from seeing them.

Hi,

Great the you are back on the `cortex debug`. I will use it in class with some 25 students this semester (starting tomorrow, running until end of March). Already today the plugin is very useful. Further improvements always welcome:)

Commenting on the "issue" and some other more or less related topics...

Issue:

I did not try combining direct `openocd` commands with the cortex-debug ITM trace. E.g., you might want to have just one channel piped to a file, but the others displayed using the plugin.

Related, On the topic of usability:

- For me each time I close and re-open a session, the `OUTPUT` console need to be opened and ITM selected in the drop down. This might be fine for casual use, but for serious development, this is a bummer. Many mouse/key presses each time, which made me actually use the raw `openocd` ITM tracing instead. Am I missing something (or is fixable, or is it a limitation of the `vscode` architecture).

- More or less each "pane" needs to be re-opened/selected each time a new debug session is restarted, so a better "memorisation" of debug layout would be nice.

- If possible we could have two buttons, one for restarting the application without recompilation/reloading, and one compile-load-restart button. If not two buttons possible (assuming the debug interface is partly hard coded in vscode) then we could have a flag defining the behavior of the "restart".

- Multiple `OUTPUT` consoles simultaneously would be great (maybe its already possible...), at least for me I can have only one. The `TERMINAL` can be split so maybe its possible.

- Assembly integration. We are programming embedded in 100% Rust, so I can't tell if it behaves better for C/C++. In any case I  never got it working showing both the disassembly and the Rust code. Also I never got it to show the disassembly when breaking. Once again, maybe I'm not using it correctly.

It works slightly better directly from the gdb prompt in the DEBUG window, but the output may be quite large, the plugin could filter out a region around the PC (like in eclipse).

Another thing that would be very useful is to have flag/config button telling whether stepping should be based on the assembly or on the source code. If you fiddle with stepping from within the `openocd` console, then you may run into inconsistency with the GUI (I mentioned that in another issue).

Somewhat related PRs:
I forked the plugin and got it to run in development mode (was quite straightforward). I applied the "preLaunchTask" patch and it worked as expected, so I think it can be released. I'm not confident enough with the internals to make a proper review of the code, but the like 3-lines altered made sense to me at least. I went through the procedures of making it an installable, it amounted to some 7mbyte. Is that expected, looks awfully big to me...

Best regards.
  Per

Marus, 2019-01-21:
@perlindgren - and regarding your question about making the installable - 7MB does seem reasonable actually. Most of this is actually related to supporting the capturing of SWO data via a serial port (v0.2.2 also updated this to work again on macOS and Linux, still working on a Windows update). That functionality requires some binary node modules - so there ends up being support files there for each supported platform. These files end up being about 1-1.5MB per platform (ia32/x64 on both Linux and Windows, and x64 on macOS) - so they do add up.

perlindgren, 2019-01-21:
@marus, aha, that explains the large size. By the way, are the SVD files also distributed with the plugin?

/Per

Marus, 2019-01-21:
Yes - there is a small selection (of mostly STM32 chips that myself and a couple of early users of the extension) use that are included and auto selected based upon the device launch parameter (if supplied).

I'm not really sure about the future of that feature actually - so haven't really been expanding it - so that also contributes somewhat in size. I'm thinking perhaps in the future perhaps it will be moved to some sort of a web service which it will use to find an SVD definition to make the repository of SVD files easier to manage (and also reduce the download size).

PhilippHaefele, 2021-10-06:
@haneefdm I think there‘s a `logfile` option for RTT. Is there also one for SWO (as i remember that they share the code base). So i think this one and #194 seem to be implemented/fixed