Marus/cortex-debug #87: JLink probe is connected, but still no debugging

tcao, 2018-08-31:
I'm in OSX 10.13.6, JLink V6.32h, VSC 1.26.1, and Cortex-Debug 0.1.21. 
When I tried to debug nRF52 DK, Adapter Output shows JLink is connected, but I still get error "Debug adapter process has terminated unexpectedly"
Any idea what's going on?
One thing I noticed the command line option passing to JLink, some are not actually supported, say "-port", or "-swoport", "-telnetport".
[image]

bikeNomad, 2018-09-18:
I have the same problem with the same versions of software. How did you get the "Adapter Output" window? I don't have one.

Travis-Spidertracks, 2018-10-24:
I got this error when arm-none-eabi-gdb was not available on my path. But I think this might be a catch-all for a number of errors

sbomber, 2018-12-18:
I experienced similar behavior.  It was because VS Code was open when I installed arm-none-eabi-gdb.  Therefore, the running instance of VS Code did not have the updated PATH.

Marus, 2019-01-20:
When you get this error can you click on the "Debug adapter process has terminated unexpectedly"  and see if it gives any additional details?

huzhanbo1996, 2019-02-07:
Has anyone solved this problem? I have the same issue. The debug will stop in 1s after it started, but it works before it stops.

huzhanbo1996, 2019-02-07:
> When you get this error can you click on the "Debug adapter process has terminated unexpectedly" and see if it gives any additional details?

my vs code give sometimes "read error", some times "exit code:-1"

closed 2019-06-28 by Marus