Marus/cortex-debug #90: Cannot read property 'uri' of undefined

GrantRolls, 2018-09-07:
I am trying to setup cortex-debug with a STM32L053 discovery board.
I have setup config as in image below, but when start debug I get the error
**Cannot read property 'uri' of undefined**
The debug config is in the user settings page as platformio overrides the local workspace settings. 

[image]

`Error: Cannot read property 'uri' of undefined
	at CortexDebugConfigurationProvider.resolveDebugConfiguration (/home/engineer/.vscode/extensions/marus25.cortex-debug-0.1.21/out/src/frontend/configprovider.js:126:77)
	at /snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:589:111
	at /snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:110:323
	at new n.Class.derive._oncancel (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:84:776)
	at Object.t.asWinJsPromise (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:110:286)
	at e.$resolveDebugConfiguration (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:589:75)
	at e._doInvokeHandler (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:671:983)
	at e._invokeHandler (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:671:701)
	at e._receiveRequest (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:670:767)
	at e._receiveOneMessage (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:670:536)
	at /snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:669:452
	at /snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:672:575
	at /snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:96:903
	at e.fire (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:98:218)
	at a (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:162:713)
	at Socket._socketDataListener (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:162:932)
	at emitOne (events.js:116:13)
	at Socket.emit (events.js:211:7)
	at addChunk (_stream_readable.js:263:12)
	at readableAddChunk (_stream_readable.js:250:11)
	at Socket.Readable.push (_stream_readable.js:208:10)
	at Pipe.onread (net.js:594:20)
log.ts:171   ERR Cannot read property 'uri' of undefined: TypeError: Cannot read property 'uri' of undefined
	at CortexDebugConfigurationProvider.resolveDebugConfiguration (/home/engineer/.vscode/extensions/marus25.cortex-debug-0.1.21/out/src/frontend/configprovider.js:126:77)
	at /snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:589:111
	at /snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:110:323
	at new n.Class.derive._oncancel (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:84:776)
	at Object.t.asWinJsPromise (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:110:286)
	at e.$resolveDebugConfiguration (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:589:75)
	at e._doInvokeHandler (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:671:983)
	at e._invokeHandler (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:671:701)
	at e._receiveRequest (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:670:767)
	at e._receiveOneMessage (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:670:536)
	at /snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:669:452
	at /snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:672:575
	at /snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:96:903
	at e.fire (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:98:218)
	at a (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:162:713)
	at Socket._socketDataListener (/snap/vscode/52/usr/share/code/resources/app/out/vs/workbench/node/extensionHostProcess.js:162:932)
	at emitOne (events.js:116:13)
	at Socket.emit (events.js:211:7)
	at addChunk (_stream_readable.js:263:12)
	at readableAddChunk (_stream_readable.js:250:11)
	at Socket.Readable.push (_stream_readable.js:208:10)
	at Pipe.onread (net.js:594:20)
errors.ts:48 WARNING: Promise with no error callback:345
errors.ts:49 
`

ad7718, 2018-09-07:
#89

ngc7293, 2018-10-31:
I had this issue, the traceback leads to 
```js
executable = executable.replace(/\$\{\s*workspaceRoot\s*\}/, folder.uri.fsPath);
```
Moving the configuration to folder settings (instead of the workspace settings) fixed. Probably an issue with folder not being set for workspaces.

PhilippHaefele, 2021-11-16:
@haneefdm Seems that the conflicting line was removed in the meanwhile 99018d7f5e644d1f11eea2aface4d8073cac7e21. So we can close that one too

closed 2021-11-16 by haneefdm