Marus/cortex-debug #91: Issue : cannot set hardware breakpoints with texane ST Link on a STM32L476

d0m72, 2018-09-12:
Hi,

I have vscode 1.27.1 and Cortex_debug 0.1.21. I'm using a discovery board from ST with a built-in STLink interface and a STM32L476 (Cortex M4). I'm using texane ST-Link 1.5.0 (recompiled). The previous ST-Link version 1.3.0 has exactly the same issue btw. The compiler is Linaro GCC 6.3 (which includes GDB).

I have this configuration for my project:

```
{
      "name": "Debug (ST-Util):  project",
      "type": "cortex-debug",
      "request": "launch",
      "servertype": "stutil",
      "cwd": "${workspaceRoot}/_obj/stm32/project/",
      "executable": "project.elf",
      "device": "STM32L476VG",
      "v1": false,
      "showDevDebugOutput":true
 }
```

What's working well is that I can do steb-by-step debugging, I can see MCU's registers, variables contents and call stack. Unfortunately, each time I set a breakpoint on my main() (or other parts of my code), the execution goes through without stopping. The program first stop on my startup code. Then if I set a break and press F5, the breakpoint is not taken into account. Of course I ensured that my code must go through the line set with the breakpoint. The location for the breakpoint is not the cause apparently as I tried several locations.

I enabled debug with **showDevDebugOutput** in my configuration file (see above) and here is an extract:

```
[...]
27-data-evaluate-expression __libc_init_array
GDB -> App: {"token":27,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["value","{void ()} 0x8004558 <__libc_init_array>"]]}}
28-data-evaluate-expression for
GDB -> App: {"token":28,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","No symbol \"for\" in current context."]]}}
29-data-evaluate-expression main
GDB -> App: {"token":29,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["value","{int (int, char **)} 0x80006b0 <main(int, char**)>"]]}}
30-thread-list-ids
GDB -> App: {"token":30,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"]]],["current-thread-id","1"],["number-of-threads","1"]]}}
31-thread-info 1
GDB -> App: {"token":31,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[[["id","1"],["target-id","Thread <main>"],["frame",[["level","0"],["addr","0x080044d0"],["func","Reset_Handler"],["args",[]]]],["file","C:\\_wp\\stm32\\stm32l476\\startup_stm32l476xx.c"],["fullname","C:\\_wp\\stm32\\stm32l476\\startup_stm32l476xx.c"],["line","114"]]]]]}}
32-stack-list-frames --thread 1 0 20
GDB -> App: {"token":32,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["stack",[["frame",[["level","0"],["addr","0x080044d0"],["func","Reset_Handler"],["file","C:\\_wp\\stm32\\stm32l476\\startup_stm32l476xx.c"],["fullname","C:\\_wp\\stm32\\stm32l476\\startup_stm32l476xx.c"],["line","114"]]]]]]}}
33-break-insert "C:\\_wp\\project\\main.cpp:50"
GDB -> App: {"token":33,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["bkpt",[["number","2"],["type","breakpoint"],["disp","keep"],["enabled","y"],["addr","0x080006ba"],["func","main(int, char**)"],["file","C:\\_wp\\project\\main.cpp"],["fullname","C:\\_wp\\project\\main.cpp"],["line","50"],["thread-groups",["i1"]],["times","0"],["original-location","C:\\_wp\\project\\main.cpp:50"]]]]}}
34-stack-info-frame --thread 1 --frame 0
GDB -> App: {"token":34,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["frame",[["level","0"],["addr","0x080044d0"],["func","Reset_Handler"],["file","C:\\_wp\\stm32\\stm32l476\\startup_stm32l476xx.c"],["fullname","C:\\_wp\\stm32\\stm32l476\\startup_stm32l476xx.c"],["line","114"]]]]}}
35-var-update --all-values global_var_Digit
GDB -> App: {"token":35,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Variable object not found"]]}}
[...]
```

I set my breakpoint on line 50 of main.cpp (which is reflected on previous report). 

The same project, compiled with the same compiler on the same target, with the same texane stlink version of gdbserver, is accepting HW breakpoints correctly under Eclipse Oxgen 3a. The only difference is the IDE, so vscode coupled to cortex-debug.

Can someone help me to understand this debug report ? Is there something special to configure so that HW breakpoints are accepted ?

Best regards.

Dominique

d0m72, 2018-09-12:
One more thing: if I manually set a hardware breakpoint when I'm in debug console output pane, I get this:

```
hbreak main.cpp:50

{"token":29,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
29-interpreter-exec console "hbreak main.cpp:50"
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Note: breakpoint 1 also set at pc 0x80006ba.\n"}]}
Note: breakpoint 1 also set at pc 0x80006ba.
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Hardware assisted breakpoint 5 at 0x80006ba: file C:\\_wp\\project\\main.cpp, line 50.\n"}]}
Hardware assisted breakpoint 5 at 0x80006ba: file C:\_wp\project\main.cpp, line 50.
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"breakpoint-created","output":[["bkpt",[["number","5"],["type","hw breakpoint"],["disp","keep"],["enabled","y"],["addr","0x080006ba"],["func","main(int, char**)"],["file","C:\\_wp\\project\\main.cpp"],["fullname","C:\\_wp\\project\\main.cpp"],["line","50"],["thread-groups",["i1"]],["times","0"],["original-location","main.cpp:50"]]]]}]}
GDB -> App: {"token":29,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
30-exec-continue --thread 1
GDB -> App: {"token":30,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"running","output":[["thread-id","all"]]}]}

```
I can see that the report tells me that I already have a (SW) breakpoint set at line 50 of main.cpp ("_Note: breakpoint 1 also set at pc 0x80006ba_".). But so far I cannot see something telling me that the HW breakpoint cannot be set.

And the results are the same : no stop.

d0m72, 2018-09-17:
I took the discovery board which I couldn't reflash with J-Link fw to another computer and I repeated all operations of ticket ([#72](https://github.com/Marus/cortex-debug/issues/72)). This time I could reflash the ST-Link interface into a J-Link. And then I modified my settings in launch.json and my own workspace settings to access J-Link's variant of GDB server. It worked like a charm this time.

In this J-Link variant I can set breakpoints wherever I want directly under vscode with my mouse and this is working correctly. So I'll switch to J-link for variant the time being. I also remarked that here it's not hardware breakpoints which are used. It's regular (software) breakpoints apparently:

```
Breakpoint 1, _GLOBAL__sub_I_assert_failed () at C:\_wp\project\main.cpp:56
56	}
```

I think that my question on previous post regarding the ST-Link with ST-Link texane GDB server is still valid. Here it would be good to understand why the breakpoints set within VSCode with the mouse (red dots in front of a line code) are not working in that case. Only HW breakpoints set manually are taken into account.

Thanks for this solution with J-link and thanks for this extension !.

Marus, 2019-01-20:
I'll have to look into the issues with Texane + ST-Link - I mostly work with and test against J-Link as that is the debug interface I tend to you myself (and I do try to keep OpenOCD). 

I expect the issue lies somewhere within Texane's st-utils itself - as it would all be using standard GDB breakpoint commands internally. 

It may also have to do with the number of breakpoints you are trying to set if you are trying to use a large number - the STM32L476VG will support 6 hardware breakpoints (and you typically want to leave at least 1 free for stepping) - so if you were trying to set too many it may have failed internally in some way that cortex-debug is not detecting properly.

closed 2019-06-28 by Marus