Marus/cortex-debug #92: Using with LiveShare

microwavesafe, 2018-09-12:
I installed the LIveShare extension recently to have a play, but it doesn't work with this extension. I get this error.
```
[2018-09-12 16:22:57.374] [renderer1] [error] Cannot convert undefined or null to object: TypeError: Cannot convert undefined or null to object
    at Function.keys (<anonymous>)
    at Function.resolveCommandValue (/home/microwavesafe/.vscode/extensions/ms-vsliveshare.vsliveshare-0.3.666/out/src/debugger/shareDebugManager.js:236:36)
    at Function.resolveCommandValue (/home/microwavesafe/.vscode/extensions/ms-vsliveshare.vsliveshare-0.3.666/out/src/debugger/shareDebugManager.js:237:63)
    at <anonymous>
```
And the debugger doesn't start. Not sure what that's about, but thought I would flag the issue. As soon as I disable LiveShare the debugger works again.

lostintangent, 2018-12-20:
@microwavesafe Which version of VS Code are you running? This issue should have been resolved in VS Code 1.30, so if you could update to the latest, as well as the latest Live Share extension, that should unblock you. Let me know if it doesn't though. If not, this would be a Live Share bug, which we can track via [our repo](http://github.com/microsoft/live-share). Thanks!

closed 2019-01-20 by Marus

lostintangent, 2019-01-20:
@marus Is this issue resolved with my instructions above? I just want to confirm that we’re doing our part to unblock users 👍

Marus, 2019-01-21:
At least with the Live Share extension just installed (VS Code 1.30.1,
cortex-debug v0.2.2, live share 0.3.1121) there doesn’t seem to be any
issues that I’ve seen, but I haven’t tried with actively using Live Share
as it isn’t an extension I’m familiar with using (although sounds
interesting). If @microwavesafe is still having issues they should open a
new ticket as appropriate.

microwavesafe, 2019-01-21:
I haven't been able to test, I will open a new ticket if I get any problems when I next try it.