Marus/cortex-debug #93: JLink not hitting breakpoints, Step Over, Step In without function, but pause working

ChiSaw, 2018-09-13:
System: macos 10.13
GNU gdb (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 7.12.1.20170417-git

SEGGER J-Link GDB Server V6.31b (beta) Command Line Version

JLinkARM.dll V6.31b (DLL compiled Mar  2 2018 17:29:41)
Firmware: J-Trace Cortex-M Rev.3 compiled Nov 23 2017 16:43:50
Hardware: V3.10
S/N: 203201163

I am trying to debug an nRF52832 with cortex-debug jusing JLink (by Segger).

As a reference I am also debugging the same code with a simulator (not using cortex-debug) where its working fine. 

I am hitting the first instruction (a function that is called) of the main(), where I set a breakpoint. It stops there. I have another breakpoint at the third instruction (also a function that is called).

When I press Continue, it is not hitting the second breakpoint (with reference simulation it is). So far it could be something goes wrong on hardware within these two. But same behavior with Step In (which should work in any case) and Step Over. With Segger Ozone (debugging tool by segger) it working just fine, but not with vs code and cortex-debug. Now it gets more strange: When I hit pause, the code actually pauses and shows me the idle while loop. 

Therefore, it seems cortex-debug is not pausing after Step In, Step Over and following breakpoints after the first one. 

It worked previously, not sure if some update introduced something. But also earlier, I experienced that sometimes it didn't hit breakpoints that where hit by Ozone on the very same code/elf file.

GDB shows

```
0x0000a062 in mbed_halt_system () at ./mbed-os/platform/mbed_error.c:68
68	            __WFI();
Not implemented stop reason (assuming exception): undefined
Resetting target
Resetting target
Breakpoint 1, main () at ./src/MYNAME/app/main.cpp:42
42	    board.setup (Icon::SUNRISE);
Program
 received signal SIGTRAP, Trace/breakpoint trap.
0x0000a062 in mbed_halt_system () at ./mbed-os/platform/mbed_error.c:68
68	            __WFI();
```

Marus, 2019-01-19:
There may be an issue with the number of hardware breakpoints the actual device supports coming in to play here. Many target devices will only actually support 2-4 breakpoints at a hardware level (and one will often need to be used for stepping through code); I don't know enough about Nordic's chips to know how many hardware breakpoints they support. In theory J-Link supports unlimited breakpoints (the do some manipulation of the program in flash) - but there may be some restrictions on that working if you are using one of the more limited J-Link devices embedded on development boards.

Other possibility I can think of is if you have heavy optimizations enabled for compiling or debug symbols not being generated properly.

RogFedex, 2020-10-15:
Observed similar issue and noticed the problem was related to callstack. limiting backtrace helped for me.
"preAttachCommands": ["set backtrace limit 6"],

haneefdm, 2020-10-16:
This is a bug with the gdb server, methinks. It should be reported to SEGGER. Closing issue

closed 2020-10-16 by haneefdm