Marus/cortex-debug #96: VSCode version 1.28 disables cortex-debug use of gdb

myxyplyxyt, 2018-10-09:
cortex-debug allowed me to debug a non-trivial application on STM32F7 discovery board using VSCode on Windows, OS X, and on Ubuntu 18.04.

It worked fine on VSCode version 1.27.2, but the VSCode team changed the debugger front end for version 1.28.

I would be willing to supply the application source code which works on 1.27.2 to try on the discovery board, if you wanted to find a work-around for the new front end on 1.28.

jonsole, 2018-10-10:
I have the same issue after upgrading to 1.28.  When debugging is started the debugger seems to get stuck with no VARIABLES shown in the sidebar and just Pause,Restart and Stop commands available.

As a workaround, opening the debug console and pressing return seems to wake up the debugger.

myxyplyxyt, 2018-10-11:
Thanks Jon,

     I tried a view things, but never considered just hitting return.

myxyplyxyt, 2018-10-11:
Workaround works just fine

closed 2018-10-11 by myxyplyxyt

0ge, 2018-10-11:
I think this is a regression and that the issue should be reopened.