Marus/cortex-debug #100: Unable to run __iar_program_start

Gigaclank, 2018-10-23:
Not sure if an issue or more lack of understanding,

[image]

Configuration as follows:
[image]

and User settings:
[image]

Unable to go to __iar_program_start therefore breakpoints not working.

0ge, 2018-10-23:
I was slowly writing a bug report (#101) while you posted this. Does the workaround work for you? (Open Debug Console and hit Enter)

Gigaclank, 2018-10-23:
> I was slowly writing a bug report (#101) while you posted this. Does the workaround work for you? (Open Debug Console and hit Enter)

Yes this works! How strange!

Gigaclank, 2018-10-23:
closing issue needed to add "runToMain": true,
close #100

closed 2018-10-23 by Gigaclank