Marus/cortex-debug #102: Cannot convert undefined or null to object

Arman92, 2018-10-23:
When I try to start the debug, VS code shows this error: 
Cannot convert undefined or null to object

My launch.json configs:
```
// {
//     // Use IntelliSense to learn about possible attributes.
//     // Hover to view descriptions of existing attributes.
//     // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
{
  "version": "0.2.0",
  "configurations": [{
      "cwd": "${workspaceRoot}",
      "executable": "./build/RigController.elf",
      "name": "Debug Microcontroller",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "jlink",
      "device": "STM32F103RE",
      "interface": "jtag",
    },
    {
      "type": "cortex-debug",
      "request": "launch",
      "servertype": "openocd",
      "cwd": "${workspaceRoot}",
      "executable": "./build/RigController.elf",
      "name": "Debug (OpenOCD)",
      "device": "STM32F103RB",
      "configFiles": [
        "target/stm32f1x.cfg"
      ]
    }
  ]
}
```

And there is no output in "Debug Console" window.
VS code version: 1.28.1

Gigaclank, 2018-10-24:
have you added your path for openocd and for armToolchainPath?
e.g.
```json
{
"cortex-debug.JLinkGDBServerPath": "C:\\Program Files (x86)\\SEGGER\\JLink_V634g\\JLinkGDBServerCL.exe",
"cortex-debug.armToolchainPath": "C:\\Program Files (x86)\\GNU Tools Arm Embedded\\7 2018-q2-update\\bin\\",
"cortex-debug.stutilPath": "C:\\Program Files (x86)\\STMicroelectronics\\stlink-1.3.0-win64\\bin\\st-util.exe"
}
  ```

does  your executable exist?

Arman92, 2018-10-25:
JLinkGDBServer and openocd were already in my PATH, however I added them in vs code settings manually:
```
{
    "cortex-debug.JLinkGDBServerPath": "/usr/bin/JLinkGDBServer",
    "cortex-debug.armToolchainPath": "/home/arman/opt/gcc-arm-none-eabi-7-2017-q4-major/bin/",
    "cortex-debug.openocdPath": "/usr/local/bin/openocd"
}
```

I'm not getting the previous error anymore, but here is a new one:
[image]

Gigaclank, 2018-10-25:
> JLinkGDBServer and openocd were already in my PATH, however I added them in vs code settings manually:
>
> ```
> {
>     "cortex-debug.JLinkGDBServerPath": "/usr/bin/JLinkGDBServer", [...]

I had this issue also. might be a stupid question is your device attached to your debugger?

Arman92, 2018-10-25:
> I had this issue also. might be a stupid question is your device attached to your debugger?

Of course!
I can even debug my target with OpenOCD, using Jlink or ST-Link. (using my own debug setup)

0ge, 2018-11-14:
I get this error when a) I have forgot to connect my debugger or 2) there is another processing using the debugger (viewing SWO, separate debug session, etc).

aguirrem, 2018-11-16:
@Arman92 did you ever get to the bottom of this?

nickray, 2018-12-07:
This issue is helpful: https://github.com/Marus/cortex-debug/issues/27

closed 2019-06-28 by Marus