Marus/cortex-debug #103: Qemu support -- new feature?

perlindgren, 2018-10-31:
Hi Folks

Has anyone looked into Qemu support? 

Maybe its already possible, just me missing it... if not perhaps something along the lines of 
https://gnu-mcu-eclipse.github.io/qemu/

Best regards
  Per Lindgren

perlindgren, 2019-01-21:
> This is a very interesting idea - will have to do some research to see how QEMU and stuff works and if it will be possible to integrate - not something I've ever worked with before so don't know enough to say what the effort will be right away.

@marus, you may have a look at cortex-m-rtfm on crates.io, there we use QEMU to simulate the scheduler (on the host). There is more info on howto use QEMU in the rust embedded book, https://github.com/rust-embedded/book/blob/master/src/start/qemu.md

Best

  Per

Marus, 2019-01-22:
I’ve got this pretty close to working last night using the information in your links - still need a couple of slight tweaks as when GDB connects things are already halted (by necessity) - so a little bit needs to be done to get the VS Code UI in sync with the state of the “device”

Marus, 2019-01-22:
@perlindgren - I made an initial preview release (manual download/install only - not on the marketplace) with initial QEMU support. Not 100% happy with it yet as occasionally (seems to be about 1/4 launches) it will seemingly not init/connect properly and the UI will think the processor is running while it is not - most of the time it does seem to work though. Restart is also a bit shaky at the moment with it not refreshing the display (and that may be a limitation as it doesn't seem to be getting the events in the GDB machine interface, but that may be a case of not sending all the correct monitor commands for QEMU).

perlindgren, 2019-01-22:
Hi Marus

What is the best way to test the "preview". Pulling the changes from your git and run the plugin in development/debug mode. Or is there a simpler way?

Best,

  Per

Marus, 2019-01-23:
There is a .visx file you can download on that GitHub release page. In the extensions panel in the overflow (...) menu at the top there is an "Install from visx" menu option - select that and select the downloaded file. You may or may not have to uninstall the marketplace version first - not sure if it will install over the marketplace one or not.

Marus, 2019-01-23:
You may not have noticed that if you were just in the email - as I forgot that before I had first clicked comment and edited to add the link at the end.

Marus, 2019-06-09:
There is a start of the QEMU support released - haven't used it a whole lot - but did a couple of quick tests that seemed to work. If you run into issues with it please open a new ticket for them.

closed 2019-06-09 by Marus